MarlinFirmware/Marlin #9963: G29 broken in v.1.1.8 latest bugfix

FlailAway, 2018-03-06:
Anycubic Linear Plus Delta/Kossel - Sensor set to trip at 0.1mm before the nozzle tip. 
1. G33 P1
2. M500
3. G28
4. G33
5. M500
6. G28
7. G29
**Expected behavior:**
Complete level and halt above last position.

**Actual behavior:**
First two lines of probing from right to left (Y to X Columns) are OK but the third line as it gets to the last X-most position but still within the probing circle, it tries to go farther Left (-X) and the stepper loses steps.

#### Additional Information
I can copy the Configuration.h Configuration_adv.h to the 1.1.8 Stable version and it uploads and performs the Bed leveling OK. But sadly the Stable version does not seem to pay any heed to the bed-leveling (M420 S1) and pushes grooves into the bed on one side, so that's why I tried the "bugfix".

Elehiggle, 2018-03-06:
Oh my god....

I have a Tevo Tarantula and have been using RCbugfix for a year now. Today I upgraded to the latest 1.1 bugfix from today. Everything was running fine. But after starting the print when it was done with the levelling (via BLtouch and Bilinear method) it was moving from the last probe point to the origin (as instructed by my starting gcode) and half way it crashes into the bed. I have been using this starting code forever and on many printers, it always worked fine. Apparently it is unrelated to me and just some bug as I see this issue being opened.

T0 ;
G21        ;metric values
G90        ;absolute positioning
M82        ;set extruder to absolute mode
M107       ;start with the fan off
G28 X0 Y0  ;move X/Y to min endstops
G28
G29
G1 Z15.0 F{travel_speed} ;move the platform up 15mm
G0 X0 Y0 Z0   ; <----------------------------------- I presume it crashes here
G92 E0                  ;zero the extruded length
G1 F200 E25              ;extrude 3mm of feed stock
G92 E0                  ;zero the extruded length again
G1 F{travel_speed}

Elehiggle, 2018-03-06:
Ok, I have done some further debugging.

I have a z offset of -10.05. After a normal Home Z/Home all, z says about 20. If I go to 0, the nozzle is at perfect height for a print/exactly leveled. If I start a bilinear bed leveling via the menu, z says 30.84 (10 higher than with Home Z for some reason). If I then go to 10 it is perfectly level, If I were to go to 0 it would crash into the bed. So I am not sure what exactly is bugged here, but it could be various things.

thinkyhead, 2018-03-07:
Whenever there are homing or leveling issues, we now ask everyone to follow a standard procedure to gather more information so that we're not just taking stabs in the dark. Here is the boilerplate:

- Download Marlin `bugfix-1.1.x` or `bugfix-2.0.x` to test with the latest code.
- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do a `G29` to probe the bed. Try not to interrupt this until you _must_.
- Copy the log output into a **.TXT file** and attach it to your next reply.

Elehiggle, 2018-03-07:
Yeah, I had Safe homing enabled and did not set XYZ of MANUAL_X_HOME_POS to 0 (the defines were commented). This causes all my issues. @FlailAway If you still have issues, follow @thinkyhead's instructions.

FlailAway, 2018-03-09:
@Elehiggle 
Thanks, but got tired of messing with Marlin and put the Smoothie board back in. The Smoothieware is crap compared to Marlin, but at least the bed leveling works out of the box. If I ever put the Trigorilla or RAMPS options back in, I'll give your suggestion a try.

Just a note on the Smoothie, it makes no difference that I can see, so if anyone reading this thinks they need a 32-bit for a delta, save your money. I was printing at 120mm/s with an Arduino-Mega+RAMPS+1.1.8, also the original Trigorilla+1.1.8 and getting excellent prints from all three. No visible difference in prints.

One print in particular included 2" of 1-1/2" internal Acme thread and if anything, the Mega+RAMPS+1.1.8 was the better of the three. Same anycubic, slicer, speed, PLA etc. Just saying.

FlailAway, 2018-03-09:
@thinkyhead 
Just a note, but I let the thing push past the limit and after rattling the steppers for a few seconds it completed the leveling. Naturally I would not use the mesh after losing steps, but I may put the RAMPS back in, in a few days and give the debug process a try for you and will post results.

I like Marlin. :)

thinkyhead, 2018-03-10:
> give the debug process a try for you and will post results

Looking forward to your report.

caffeineflo, 2018-03-11:
@thinkyhead I head about the same issue with a FW that was about 24hrs old (that is, last time I merged bugfix/1.1.x). I just pulled again and at least the issue described here is gone.

One thing that is still strange though, at the end of probing, Z will be set to Z_CLEARANCE_BETWEEN_PROBES while it was set to Z_CLEARANCE_DEPLOY_PROBE before (which was way better for me, as I have to remove the probe after probing)

thinkyhead, 2018-03-11:
@caffeineflo — Thanks for reminding me to have some caffeine. And also, for your report!

The issue is that we tweaked the raise in `set_probe_deployed` so that `FIX_MOUNTED_PROBE` only does a raise during deploy, but for stow it doesn't do any raise. This was a change by request. I think it was probably in relation to the case where the probe is the nozzle, which is also a case of `FIX_MOUNTED_PROBE`. Some users wanted the nozzle to remain at the bed level.

Seems like it would be a good idea to add the option `Z_AFTER_PROBING` to give users like yourself more control over the final position.

thinkyhead, 2018-03-11:
@caffeineflo — Please test the `Z_AFTER_PROBING` option that I've added in [this branch](https://github.com/thinkyhead/Marlin/archive/bf1_z_after_probing.zip). It applies to all probe procedures, including `G28` when the probe is used, `G30`, and `M402`.

caffeineflo, 2018-03-11:
@thinkyhead I've just pulled in and tested your changes, works for me as expected! I like it a lot as I got a nozzle probe that you attach directly to the nozzle and remove after you're done! 

Thanks a lot!

thinkyhead, 2018-03-11:
> nozzle probe that you attach directly to the nozzle and remove

It was easy to build by considering that use-case, but it will also be useful in-general. Many users would prefer that the nozzle is lifted after probing, especially when the nozzle is the probe.

closed 2018-03-11 by thinkyhead