MarlinFirmware/Marlin #9967: M420 data explanation please

ruggb, 2018-03-07:
what does this data tell me from M420 V command?

Bed Level Correction Matrix:+1.000000 +0.000000 +0.000587
+0.000000 +1.000000 -0.000377
-0.000587 +0.000377 +1.000000

AnHardt, 2018-03-07:
https://en.wikipedia.org/wiki/Rotation_matrix

ruggb, 2018-03-07:
I have already looked at that. It is a little heavy for my brain. Can you tel me in 1st yr college terms. My latest one is
+1.000000 +0.000000 -0.000002
+0.000000 +1.000000 +0.000004
+0.000002 -0.000004 +1.000000

is this better or worse for a 3 point level

AnHardt, 2018-03-07:
Position vectors are multiplied with this matrix to get the corrected positions.
A ideally level bed has the matrix
```
1 0 0
0 1 0
0 0 1
```
Larger differences to this mean larger tilt. -> Your bed is verry level.

closed 2018-03-07 by thinkyhead