MarlinFirmware/Marlin #9972: [bugfix_2.0.x] Filament Runout

bwgeez02, 2018-03-07:
Arduino IDE 1.9 beta won't compile due to fil_runout_pin error. Pin is not defined in pins_RADDS.h. I added this code, from Wurstnase Marlin4Due, to Misc. Functions and changed FILRUNOUT_PIN to FIL_RUNOUT_PIN and it compiled. 

#if defined(FILAMENT_RUNOUT_SENSOR)
  #define FIL_RUNOUT_PIN        -1
#endif

bwgeez02, 2018-03-07:
Also works with just 

#define FIL_RUNOUT_PIN     -1

And of course a pin assigned.

thinkyhead, 2018-03-07:
Please post the error message. We'd like to allow Marlin to compile successfully with the filament 
runout sensor disabled, regardless of whether `FIL_RUNOUT_PIN` is defined.

bwgeez02, 2018-03-07:
It compiles if FILAMENT_RUNOUT_SENSOR is commented. Uncommented and this error is received 

Arduino: 1.9.0-beta (Windows 7), Board: "Arduino Due (Programming Port)"

In file included from sketch\src\HAL\HAL_DUE\../../inc/MarlinConfig.h:34:0,

                 from sketch\src\HAL\HAL_DUE\EepromEmulation_Due.cpp:35:

sketch\src\HAL\HAL_DUE\../../inc/SanityCheck.h:447:6: error: #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."

     #error "FILAMENT_RUNOUT_SENSOR requires FIL_RUNOUT_PIN."

      ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Why not just add the FIL_RUNOUT_PIN to pins_RADDS.h?

thinkyhead, 2018-03-09:
> Why not just add the FIL_RUNOUT_PIN to pins_RADDS.h?

You'll need to tell us what pin to use, so we can impose that on everyone else.

bwgeez02, 2018-03-10:
Haha. I use the max z endstop for my filament sensor so pin 38. I'm trying to learn github so I can help out more. I appreciate everything everyone does for the open source community.

thinkyhead, 2018-04-01:
Well, clearly we can't presume to enforce that pin. It would make no sense for Delta users and others with a Z Max endstop. The Z Max endstop is also the first pick for the Z Probe pin (when not assigned to Z Min). Only a few boards have a pin set aside, I would guess because that pin is already documented as the commonly-used pin for certain 3D printer model(s). I see that `pins_RAMPS.h` has picked the Servo 3 pin, since that's the most likely to be free. If RADDS has such a pin, that might make the best default.

closed 2018-04-01 by thinkyhead