MarlinFirmware/Marlin #9980: Motor power timeout during G29 P2

golobodo, 2018-03-07:
### Description

During the manual probing I realized, that especially the Z-Motors disables after DEFAULT_STEPPER_DEACTIVE_TIME. The motor slightly move, but haven't enough power to move the Z-Axis -> step losses. I'm able to rotate the Z threaded rod by hand without force. This may affect all axes. 

### Steps to Reproduce

0. Using latest bugfix_1.1.x
1. M502
2. M500
3. M501
4. M190 S60
5. M104 S210
6. G28
7. G29 P1
8. G29 P2 B T
9. wait DEFAULT_STEPPER_DEACTIVE_TIME, in my case 240 seconds

**Expected behavior:**
Motors being active during manual probing

**Actual behavior:** [What actually happens]
Motors disabled after DEFAULT_STEPPER_DEACTIVE_TIME

#### Additional Information
An workaround is to increase DEFAULT_STEPPER_DEACTIVE_TIME or maybe setting DISABLE_INACTIVE_Z to false

Roxy-3D, 2018-03-07:
Thank you for finding this problem.  This should not be too hard to fix.  But the problem is I don't have a machine available to test any changes right now.    I think we just need to make
```
void manually_probe_remaining_mesh();
```
suspend the time out on each mesh point....

thinkyhead, 2018-03-09:
> <---<<<  This line is new.

Try out ` ```diff ` when starting a code block. It colorizes lines starting with `-` and `+`. Very cool.

golobodo, 2018-03-09:
Adding the line solved the problem. You can add it to the source. Thanks!

Roxy-3D, 2018-03-10:
Oh!   OK!   That is a nice feature...

closed 2018-03-10 by thinkyhead

VanessaE, 2018-03-20:
Is this related:

I was re-tuning my E steps/mm after replacing a worn-out hob gear, and as you no doubt are aware, one is supposed to mark the filament, very slowly (in this case, 50 mm/min) extrude 100 mm measure how much was really moved, adjust E as needed, and repeat until it's right.

At the end of each of those 100 mm extrude-only moves, the extruder bounces backward from the pressure, and I can rotate it by hand easily.  My idle timeout is 120 seconds, which is exactly how long each of those extrude moves takes.

thinkyhead, 2018-03-21:
Doing that from the LCD @VanessaE ? I'll add a `refresh_cmd_timeout` to the `_lcd_move` routine and that should fix it.

VanessaE, 2018-03-21:
Nope, from Pronterface (I'm beginning to think I'm the only one whose bot lacks an LCD :stuck_out_tongue: ), though I imagine both end up doing the same thing.

thinkyhead, 2018-03-21:
I'm guessing the issue is that the stepper timeout isn't respecting that a move is in the buffer as it counts down. It should reset itself in response to any move. I've never liked the fact that Marlin has to use `refresh_cmd_timeout` to prevent steppers from timing out, rather than resetting any time a moves comes into the planner.

VanessaE, 2018-03-21:
I was thinking along the same lines.

thinkyhead, 2018-03-21:
I see that the timeout routine is checking whether there's a block queued in the planner. However, it's not resetting the countdown. As soon as the long move completes it disables the steppers. I'll fix that now.

thinkyhead, 2018-03-21:
The PR #10178 should fix things. Please give it a try whenever you have a chance.