MarlinFirmware/Marlin #9983: E2END - EEPROM Size value missing in pins_RAMPS_SMART.h

LichtiMC, 2018-03-07:
Using a SMART_RAMPS Board when compiling marlin with UBL defined it errors out with error "'E2END' was not declared in this scope"

SMART_RAMPS Board uses the 24LC256SN I2C eeprom.
So there should be a line 
´´´#define E2END 0xNNNN```

I don't know which values to take here. 24LC256SN is 8 x 32KB = 256KB eeprom.
32KB would be 8000 in HEX, but there were eeprom-errors when using this value.

Please tell me the right value or add it to file pins_RAMPS_SMART.h

p3p, 2018-03-07:
For a 32KB (8 x 32Kb) EEPROM the E2END value would be 0x7FFF, but that's far larger than usual, if that doesn't work you could try with a value of 0xFFF as that's more in the realms of what is expected (standard ATmega2560 value).

LichtiMC, 2018-03-08:
Thanks for the answer.
It only works with 0xFFF. 

I also tried 0x2000 but then I couldn't save the mesh to Slot 0 and Slot 1. I could save it to slot 17 though...!? I guess there's something odd with bigger eeproms for now...

thinkyhead, 2018-03-10:
Thanks! Added to the Smart RAMPS pins and made sure that E2END is set correctly in other cases.

closed 2018-03-10 by thinkyhead

Blaster1920, 2018-05-15:
I'm experiencing the same problem with an Alligator R2.
How can i fix it?

thinkyhead, 2018-05-16:
> How can i fix it?

Add to the `pins_ALLIGATOR_R2.h` file:
```cpp
#define E2END 0x1FFF
```