MarlinFirmware/Marlin #10000: Marlin has 10,000 threads!!!

Roxy-3D, 2018-03-08:
Wow!   10,000 threads!    At:  www.reprap.org/wiki/List_of_Firmware   They say in the description of the Marlin section:
```
Development on this firmware appears to be very active.
```

That is very true...

Grogyan, 2018-03-08:
Congrats on this massive milestone.

Just shows how active the development is.

thinkyhead, 2018-03-08:
Of course, most of them are threads we can… ignore.

Studiodyne, 2018-03-08:
Like this !
😄

Studiodyne, 2018-03-08:
Question : 
Is there a tool that can add our own feature in each new marlin release
Because , many here have made their own code , and each time we want to try all new release , IT S A BIG ADVENTURE in notepad++

Is there an auto tool that can add in one click , all 

Thks

Spawn32, 2018-03-08:
BTW ! Congrats !!! is there something like a virtual cake in her ? 😄

Roxy-3D, 2018-03-08:
> Of course, most of them are threads we can… ignore.

Yes...  Of course.   But also remember, 1/2 of them are Pull Requests with changes to the code base.

10,000 is a serious number.   And some of the threads are very important (and old).   #7076 for example.    That was (almost) 3,000 thread creations ago, and it is still active.

Anyway...   I thought it was worth pointing out we hit the 10,000 number.

Elehiggle, 2018-03-08:
@Studiodyne No. But you can create a fork of Marlin and auto merge new updates. That is the way to go to achieve what you want.

Studiodyne, 2018-03-08:
@Elehiggle  thanks ! 😄

Roxy-3D, 2018-03-09:
> @Studiodyne No. But you can create a fork of Marlin and auto merge new updates. That is the way to go to achieve what you want.

And when you do that...   It is helpful to try to not have a bunch of 1 line code changes everywhere.  If you can just insert a block of code in one place it will help you not to have 'conflicts' with the merge process.  That isn't really possible.   But the more you can go in that direction the less likely it is you will have problems pulling updates into your fork.

Stated another way with different words...   As long as you are not making changes to the same lines of code the updates are changing, it is very easy to pull in the updates.   But if you are making changes to the same lines of code the updates change, you will have a 'conflict'.

Studiodyne, 2018-03-09:
@Roxy-3D   and @Elehiggle 
THANK YOUUUUUUUUU
I just discover '  compare base fork ' , and i merge all news in my fork
My features are 100% features and have a special fork , it's why i will never have conflicts
and for the fixes that i can make for contributing , i make a special fork too

Question of git maths
(My old base fork + my features) + merge the last updated base marlin = New updated base + my features
all my features are not included in the comparaison , only the bases are compared 
Exact ?
I have made an update ,and my features still alive  , i think GIT is simply wonderfull !!!

marcio-ao, 2018-03-09:
That's pretty impressive for a MCU. I thought it could support a few threads at most.

JK.

Studiodyne, 2018-03-09:
@Roxy-3D @Elehiggle 
Thk you very much , it's a really good news i've waited so long
I have made a new fork , and i've add all my feature , the complete marlin for my machine
So now , if i want to update my fork , i just have to merge ' From marlin  to my fork ' , and all updates will be merged , without deleting my feature . Only the head forks are updated , and my features are keeped
Is it exact ?
I can make it , when i click on ' new pull request ' but with option ' compare head forks ' and i merge the marlin original upstream to my own
I cannot made it today , but  everytime i need to have an update , i will make it

All is exact ?

thinkyhead, 2018-03-10:
> So now , if i want to update my fork , i just have to merge ' From marlin to my fork ' , and all updates will be merged , without deleting my feature .

Merge works. But [rebase is better](https://hackernoon.com/mastering-git-why-rebase-is-amazing-a954485b128a).

Studiodyne, 2018-03-10:
@thinkyhead 
Thks so much , i have tested and it's great ,  i have made a new fork and add my feature , i've seen you have made a commit , i've tested it and i see github update the ' head fork ' and add my feature after
It's a dream for me !!! 
Not you're talking about rebase .... and i have searched what it means , now , i have more experience and i'm sure i will quickly understand how to 
Because the ' word ' rebase exists nowhere in github website

Anytime , someone talk about git functionnality , he gives only the ' command line ' , but i hate command line and i use only the website , and i don't know where is ' REBASE BUTTON ' .

Where is the rebase command on this website ... ?

Studiodyne, 2018-03-10:
Yes i have found it
I create a pull request  from upstream to my branch
and the big green button , have a pop list , and this pop list , i have ' squash , merge , and rebase '
Yes

Thksssssssssss

Studiodyne, 2018-03-10:
@thinkyhead 
Thks a lot , i begin to understand all 
Can you help me , on a few questions please

1- The rebasing ,  rebase my branch with my ' own 2.0 bugfix ' , or the marlin upstream 2.0 bugfix that is the source of my fork ?
2- If the rebase , use only my own master , i need to update it by merging the upstream before ?
3-) Or may be , git is smart , and when i ask for rebase , it will update all my bases on my fork 

I don't know

thinkyhead, 2018-03-10:
Conflicts are a fact of life, regardless of whether merge or rebase is used. Resolving them is an art.
```
git checkout my_feature_branch
git fetch upstream
git rebase upstream/bugfix-2.0.x
```

Studiodyne, 2018-03-10:
thk you man !!!!!!!!

Please can you have an eye on my Project here , read all and ... love it please ... impossible for me to make it without help
at the end
[2085](https://github.com/MarlinFirmware/Marlin/issues/2085)

lrpirlet, 2018-03-10:
Hi @Studiodyne 
Maybe you can help yourself... Beside trying **github training**, **github documentation** and the like in google, you should get hold of the official manuals under https://git-scm.com/doc (yes, you can try google **github manual**)

Note: Tu trouveras même en français le meilleur de la documentation à propos de GIT et GITHUB sous ce lien: https://git-scm.com/book/fr/v2
J'ai suivi un cours gratuit trouvé sur la toile. En quelques heures j'ai compris la philosophie de GIT et GITHUB (le pourquoi) avec bien assez de connaissance pour maitriser le comment... (non, jai perdu la référence désolé). Cependant, un peu de recherche systématique devrait porter ses fruits.
Bonne chance...

Studiodyne, 2018-03-10:
@lrpirlet 
Merci bcp , en fait , Github n'est pas si compliqué , mais compliqué a comprendre lol , j'ai quand meme fait plusieurs recherches et tutoriels , mais a chaque fois ... c'est  ligne de commande et pas ' navigator ' , la j'ai bien avancé , je suis capable de garder mes features , et de mettre a jour la base , et ca me suffit.
En fait en qq heures ici sur ce topic , j'ai bien maitriser le système , bien plus qu'en qq mois ici , a bricoler par flemme de m'y attaquer. Pour accélérer et éviter les tutos , j'ai même fait des recherches locales pour trouver qq'un qui me forme en qq heures , mais personne sur Marseille n'a jamais répondu ...
Maintenant je pige , git hub , est intelligent , il sait garder la base , et les ajouts  et ça permet de mettre a jour la base sans écraser les ajouts , une fois qu'on a compris ça , tout s'illumine lol
Merci pour ton aide ..
Cdlt

marcio-ao, 2018-03-12:
@thinkyhead : Merge works. But rebase is better.

I'm somewhat of git n00b as well, so bear with me. I've heard that rebases are bad because they rewrite history and the advise is that if commits are made public, then you should not use rebase. However, it seems like if you are collaboratively developing on github, then your commit are by default being pushed to your public repo. Wouldn't in this case the advise to rebase go against the prohibition against rebasing branches that are publically available?

thinkyhead, 2018-03-12:
> rebases are bad because they rewrite history and the advice is that if commits are made public, then you should not use rebase

It depends on how you use it.

The way we work is that someone makes a fork of Marlin, creates a copy of some working branch (either `bugfix-1.1.x` or `bugfix-2.0.x`) and then adds commits to their working copy. They may submit that branch in the form of a pull request, targeting the branch that they started with (`bugfix-1.1.x` or `bugfix-2.0.x`).

So, you have a PR, and while it's sitting and waiting for review, more changes are coming in to the branch you're targeting. So after a while conflicts arise, and the working branch starts to get stale.

You have two options:
- Merge the additional changes from the upstream bugfix branch, which brings in all the new commits from upstream, appending them after your feature commits. If you are required to resolve conflicts, those resolutions are collected in the additional "merge commit."
- Rebase your commits onto the updated bugfix branch that you're targeting. This replaces the whole branch up to (not including) your commits, and then applies your commits at the end. If there are merge conflicts, they are made in-place and alter your commits.

Note that in both cases, you have only modified your working copy of your feature branch. At this point you have not pushed the updates to your fork on Github.

When you finally do push your updated branch, the result differs depending on whether you merged or rebased.

If you merge regularly, you continue to add on more "merge commits" that clutter up the PR, and which give you something like this:
```
  mystuff1 > mystuff2 > bugfixstuff > bugfixstuff > myfixes1 > bugfixstuff > myfixes2
```
Following the changes is difficult in a PR that has been composed this way.

If on the other hand, you rebase and use `git push -f` to update your branch in your fork, you get something much nicer:
```
  mystuff1 > mystuff2
```
It is much easier to see the changes in this PR, which is consolidated and composed only of changes that would be made to the latest `bugfix-1.1.x` (for example). You don't have to read through the "commits that applied 2 weeks ago" followed by "the new commits to do it the new way" and "commits that resolve conflicts," &c.

You may also use interactive rebase (`git rebase -i upstream/bugfix-1.1.x`) to clean up your history, combine commits, re-order them, etc. And when you `git push -f` those up to your Github account, they completely replace the commits previously appearing in the PR. It looks shiny and new, and it is.

I would be happy to demonstrate the differences in a live video hangout sometime, along with the tricks and scripts I use to streamline the process. It might also help to demonstrate how to deal with conflicts, whether using merge or rebase.

marcio-ao, 2018-03-12:
`And when you git push -f those up to your Github account, they completely replace the commits previously appearing in the PR. It looks shiny and new, and it is.`

So, the cardinal rule about not rewriting history is in fact violated, but I guess this is considered okay in this case due to the benefits and the fact that is is a special pr branch that you're hoping nobody checked out? If someone had checked out your PR branch, then they would be affected once your force pushed the rebased branch, correct?

`I would be happy to demonstrate the differences in a live video hangout sometime, along with the tricks and scripts I use to streamline the process. It might also help to demonstrate how to deal with conflicts, whether using merge or rebase.`

That may be useful for other contributors as well. Maybe not just a live video, but something which is recorded and posted somewhere for posterity :)

thinkyhead, 2018-03-12:
> So, the cardinal rule about not rewriting history is in fact violated

Nothing is "violated." It has no effect on the history of `MarlinFirmware/bugfix-N.N.x`, but only on your working copy, and that is just a piece of scrap paper.

thinkyhead, 2018-03-12:
How often do you do this important thing to keep your fork up to date?
```
git fetch upstream
git checkout bugfix-1.1.x
git reset --hard upstream/bugfix-1.1.x
git push -f origin bugfix-1.1.x
```

thinkyhead, 2018-03-12:
And then there's `git diff`… Another way to get out of painting yourself into a corner…

marcio-ao, 2018-03-12:
> Too many Github users get so stuck and confused because they don't maintain their forks, and simply end up deleting them and replacing them, and Github forces them to choose a new name for their fork, slightly breaking its association with the original project.

Yes, I've done this. Guilty as charged :)

> Nothing is "violated." It has no effect on the history of MarlinFirmware/bugfix-N.N.x, but only on your working copy, and that is just a piece of scrap paper.

I guess I don't understand GIT well enough to grok this. I thought that once I created my branch, made changes, and then pushed it to my fork on Github to create the PR in upstream, then it was made public and any rebase after that point would affect others adversely.

marcio-ao, 2018-03-12:
> No shame in rewriting history or outright git resetting in the name of keeping your fork in sync and intact.

Yes. I suppose that makes sense. Anyone who looks at a fork on github is probably begging for trouble anyway. Like you said, the majority of forks are created for a pull request and then ignored. It seems like GItHub ought to come up with a better method, like a temporary fork that expires when all PRs are closed.

thinkyhead, 2018-03-12:
> any rebase after that point would affect others adversely.

We only rarely have PRs submitted to branches which are themselves PRs. But, in that case any change to your branch, whether due to a merge or rebase, can adversely affect that second-level PR. Fortunately, once someone submits a PR to one of your branches, you gain the permission (by default) to freely modify that branch too. I take full advantage of that privilege. 🍷

closed 2018-03-16 by thinkyhead