MarlinFirmware/Marlin #10001: Unable to get full file name on serial port

spilz87, 2018-03-08:
hello,

I don't know if it's an issue or if it's a bad configuration on my marlin, but I'm not able to get the full file name on serial port : i get only the 6 first letters like "abcdef~1.gco" 

how to reproduice : 
- set gcode file on SD card
- start print from SD card with LCD
- read serial port output

is it normal ?
is there a way to get the full name ?

thanks for your help

regards

TheSFReader, 2018-03-08:
Looks like M33 is the command you're loking for : yous hould be able to feed it the "MSDOS Compatible path", and retrieve the full path cf. http://marlinfw.org/docs/gcode/M033.html (requires LONG_FILENAME_HOST_SUPPORT )

spilz87, 2018-03-08:
I saw this command, but if I understand it well, I need to know the path and file name before :
In my exemple I have to send "M33 abcdef~1.gco" to get "abcdefghijkl.gcode", right ?
there is no way Marlin send directly the full name when print start ?

TheSFReader, 2018-03-08:
You're right, there doesnb't seem to be any  way to get it in that case. Feature Request probably (doesn't _seem_ too hard to at least add an option to get the full path and name of the current printing file, bu will have to check.)

thinkyhead, 2018-03-09:
If the print is initiated from a host, then the host knows the path to the file and can use `M33` to obtain the long name of the file. If the print is not initiated from a host, then —generally speaking— hosts are not meant to interact with SD printing.

spilz87, 2018-03-09:
Actually I'm building a project with esp8266 "spying" the serial port and get all information on a web page, it's not really a host, so I don't want to interact with SD printing (except emergency stop), just get all information about the print as possible 
The goal is to add a camera and check if printing is ok (stop if not)

thinkyhead, 2018-03-11:
Relocating topic to #10055

closed 2018-03-11 by thinkyhead