MarlinFirmware/Marlin #10002: Lockups after moving axes

livid-fpv, 2018-03-08:
I've got a really unusual issue with a corexy i've just finished. 
I'm using an mks gen 1.4 that is known good as i used in another machine, watterott tmc2130's wired up correctly, m122 shows good results (address for all the drivers is the same, is that normal?). Full graphic smart controller (I remapped the CS pins to get them both working at the same time). The printer boots, my software controlled LED's work fine, and the axes' sort of work, which is where my issue resides
If i move the axes around through either pronterface, simplify or the controller, it works for a bit, then the entire controller locks up and the serial activity LED's go mental, even when not connected to my laptop. Pressing the estop on the controller doesn't fix it, neither does sending a reset command over the PC. I have to unplug the entire machine from the wall and my pc to stop it. any ideas on how to fix it?
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1792756/Configuration.zip)

It's related to moving the axes around as if i don't, and fiddle with the heaters or case lights for the same amount of time the machine is fine

I'll get a video of the issue once i get back to the machine in an hour or so

Studiodyne, 2018-03-08:
I have a gen 7 v1.4
Here is my pinout 

/**
 * Gen7 v1.4 pin assignments
 */
```

#if !defined(__AVR_ATmega2560__) && !defined(__AVR_ATmega644P__) && !defined(__AVR_ATmega644__) && !defined(__AVR_ATmega1284P__)
  #error "Oops!  Make sure you have 'Gen7' selected from the 'Tools -> Boards' menu."
#endif

#define BOARD_NAME "Gen7 v1.4"

#define GEN7_VERSION 14 // v1.4

#define KNOWN_BOARD 1

#define LARGE_FLASH        true

#define X_STEP_PIN         54
#define X_DIR_PIN          55
#define X_ENABLE_PIN       38 
#define X_MIN_PIN           3
#define X_MAX_PIN          -1 // 2

#define Y_STEP_PIN         60
#define Y_DIR_PIN          61
#define Y_ENABLE_PIN       56
#define Y_MIN_PIN          14 
#define Y_MAX_PIN          -1 // 15

#define Z_STEP_PIN         46
#define Z_DIR_PIN          48
#define Z_ENABLE_PIN       62
#define Z_MIN_PIN          18
#define Z_MAX_PIN          -1 // 19

#define Z2_STEP_PIN        36
#define Z2_DIR_PIN         34
#define Z2_ENABLE_PIN      30

#define E0_STEP_PIN        26
#define E0_DIR_PIN         28
#define E0_ENABLE_PIN      24

#define E1_STEP_PIN        36
#define E1_DIR_PIN         34
#define E1_ENABLE_PIN      30

#define SDPOWER            -1
#define SDSS               53
//#define LED_PIN            -1

#define FAN_PIN            9// D8 for dual extruder 131227

#define PS_ON_PIN          12

#define FIL_RUNOUT_PIN 15
#define FIL_RUNOUT2_PIN 2

//#define LASER_PIN   -1

#define HEATER_0_PIN       7   // EXTRUDER 1
#define HEATER_1_PIN       10    // EXTRUDER 2 (FAN On Sprinter)

#define TEMP_0_PIN         3   // ANALOG NUMBERING
#define TEMP_1_PIN         4 //14   // ANALOG NUMBERING

#define HEATER_BED_PIN     8    // was 14 or 15 BED
#define TEMP_BED_PIN       14   // ANALOG NUMBERING

#define SERVO0_PIN         11
//#define SERVO1_PIN         -1
//#define SERVO2_PIN         -1
//#define SERVO3_PIN         -1

#ifdef ULTRA_LCD
  #ifdef NEWPANEL
       #define BEEPER_PIN 37
              #define DOGLCD_A0  27
       #define DOGLCD_CS  25
       #define LCD_BACKLIGHT_PIN 65  // backlight LED on A11/D65
       #define SDSS   53
       
       #define KILL_PIN 64
       // GLCD features
       //#define LCD_CONTRAST 190
       // Uncomment screen orientation
         //#define LCD_SCREEN_ROT_90
         //#define LCD_SCREEN_ROT_180
         //#define LCD_SCREEN_ROT_270
       //The encoder and click button
       
       #define BTN_EN1 31
       #define BTN_EN2 33
       #define BTN_ENC 35  //the click switch
       #define SD_DETECT_PIN 49
    #endif
#endif //ULTRA_LCD
```

livid-fpv, 2018-03-08:
I used the ramps 1.4 target, but like i said the board worked perfectly on a cartesian machine with 8825 drivers also configured as a ramps

Studiodyne, 2018-03-08:
Ok ,it's a core xy ... i can't help

8Roman8, 2018-03-08:
I had a similar problem with my printer, if I move the z axis too fast the controller stops and I have to turn the power off to reset. I adjusted the Max speed and this solved the problem. Maybe your controller have to send too much steps per second?

livid-fpv, 2018-03-08:
you know what, it might be that. i've got some fairly high mm/s and steps per mm configured as i'm running 256 microstepping, i'll swap out to the latest bugfix build and turn down the speeds. if that doesn't work i'll also turn down the microstepping

livid-fpv, 2018-03-08:
seems to only be on auto home now

8Roman8, 2018-03-08:
Marlin can handle about 40khz step rate, by native 256 microsteps you can run a 200 steps per revolution motor with only ca. 45RPM and I'm sure you want to run your printhead faster than that.
With atmega2560 I would run MAX 32 microsteps per step.

livid-fpv, 2018-03-09:
ok. i've gone down to 32 microsteps at 500mm/s, deeply regretting not choosing a duet board for this build now. duet is so easy

8Roman8, 2018-03-09:
500mm/s hears too much for me, how many steps/mm du you have? 

If you have 80steps/mm so your board must send a steprate of (32microsteps/step × 80steps/mm × 500mm/s) about 1.3MHz... 
Marlin on a 8bit processor can only handle 40kHz... 

And I hope you know that a higher microstepping above 16 don't give you a better accuracy... (a standard stepper motor have a step accuracy of 5%)

livid-fpv, 2018-03-09:
I know, but, but higher numbers!!! lol, ok. i'm using 80 steps per mm at 32, should i try 200?

8Roman8, 2018-03-09:
Sry, I have made a mistake in my calculation ^^

80x500=40kHz... (without microsteps)

Also 500mm/s should work fine.

thinkyhead, 2018-03-09:
> i'm using 80 steps per mm at 32, should i try 200?

32x micro-stepping might be a little quieter than 16, but 16 is usually "good enough." For the Z and E axes you can go down to 8x micro-stepping without worry. Of course, your steps-per-mm value is determined entirely by your micro-steps, pulley teeth, belt pitch, and stepper motor. (e.g., 16 x 20 x 2 x 200).

livid-fpv, 2018-03-09:
I want the machine to be as quiet as possible. i noticed going to 32 broke the z axis, i had to take it back to 256 to get it working again

8Roman8, 2018-03-09:
arent you using TMC2130? 
you can run your motor driver with 16 / 32 microsteps and activate the interpolation....

dont forget to M502 after flashing, to load your new steps per mm , maybe this broke your z axis?

livid-fpv, 2018-03-09:
No, I adjusted the steps per mm correctly, when i tried running the z axis it made a buzzing sound and didn't run smooth. I have interpolation enabled. on a machine with a bed as small as mine (150x200) 500mm/s is fairly insane anyway, i'll drop it to 250

thinkyhead, 2018-03-09:
Once you find settings that your processor can handle without blowing up you can close this issue.

8Roman8, 2018-03-09:
If you adjust pteps per mm im the arduino IDE and upload the new fw, the old steps per mm are still saved in the EEPROM, you have to M502 to load your fresh uploaded settings, otherwise your old steps/mm are used by the controller and your axis don*t move because the speed is too high with lower microsteps.

livid-fpv, 2018-03-09:
Lol ok thanks. Like i said i'm not much used to marlin. I'll give it a shot later, i might go up to 64 and half the speed

8Roman8, 2018-03-10:
Have you get your machine working, I think you can close the issue?

livid-fpv, 2018-03-10:
I haven't got everything working, but this issue seems solved

closed 2018-03-10 by livid-fpv