MarlinFirmware/Marlin #10003: M122 etc. serial output broken

8Roman8, 2018-03-08:
bugfix 2.0.x
[configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1792895/configuration.zip)

this is the output after M122 or M911:

> Send: M122
> Recv: 		=�N�_�)�)�:�K�\��������� ��\x1c#�\x10�`�H\x12p�I\x12��J\x12��K\x12��8\x12��9\x12��:\x12��;\x12\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	=�N�_�)�)�:�K�\��������� ��\x1c#�\x10�`�H\x12p�I\x12��J\x12��K\x12��8\x12��9\x12��:\x12��;\x12\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	=�N�_�)�)�:�K�\��������� ��\x1c#�\x10�`�H\x12p�I\x12��J\x12��K\x12��8\x12��9\x12��:\x12��;\x12\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	=�N�_�)�)�:�K�\��������� ��\x1c#�\x10�`�H\x12p�I\x12��J\x12��K\x12��8\x12��9\x12��:\x12��;\x12\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��
> Recv: Enabled		false	false	false	false	false	false
> Recv: Set current	950	950	600	600	600	600
> Recv: RMS current	939	939	581	581	581	581
> Recv: MAX current	1324	1324	819	819	819	819
> Recv: Run current	16/31	16/31	18/31	18/31	18/31	18/31
> Recv: Hold current	8/31	8/31	9/31	9/31	9/31	9/31
> Recv: CS actual		8/31	8/31	9/31	9/31	9/31	9/31
> Recv: PWM scale	0	0	0	0	0	0
> Recv: vsense		0=.325	0=.325	1=.18	1=.18	1=.18	1=.18
> Recv: stealthChop	true	true	true	true	true	true
> Recv: msteps		16	16	16	16	16	16
> Recv: tstep		1048575	1048575	1048575	1048575	1048575	1048575
> Recv: pwm
> Recv: threshold		94	94	82	82	164	168
> Recv: [mm/s]		84.11	84.11	3.01	3.01	30.02	30.07
> Recv: OT prewarn	false	false	false	false	false	false
> Recv: OT prewarn has
> Recv: been triggered	false	false	false	false	false	false
> Recv: off time		5	5	5	5	5	5
> Recv: blank time	24	24	24	24	24	24
> Recv: hysterisis
> Recv: -end		2	2	2	2	2	2
> Recv: -start		3	3	3	3	3	3
> Recv: Stallguard thrs	2	2	8	8	0	0
> Recv: DRVSTATUS	=�N�_�)�)�:�K�\��������� ��\x1c#�\x10�`�H\x12p�I\x12��J\x12��K\x12��8\x12��9\x12��:\x12��;\x12\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	=�N�_�)�)�:�K�\��������� ��\x1c#�\x10�`�H\x12p�I\x12��J\x12��K\x12��8\x12��9\x12��:\x12��;\x12\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	=�N�_�)�)�:�K�\��������� ��\x1c#�\x10�`�H\x12p�I\x12��J\x12��K\x12��8\x12��9\x12��:\x12��;\x12\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	=�N�_�)�)�:�K�\��������� ��\x1c#�\x10�`�H\x12p�I\x12��J\x12��K\x12��8\x12��9\x12��:\x12��;\x12\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03��	\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c�� [... cut ...]
> Recv: 	\x0c�q!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c�\x1b�\x0c��!\x0c��!\x0c��!\x0c���\x0c��!\x0c��!\x0c��!\x0c��!\x0c�=�\x0c�Z�\x0c��!\x0c��:\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!\x0c��!UW`WgWmWsWyW\x7fW�W�W�W�W�W�WzX�X�X�X�X�X�X�X�Y�X�Y\x06Y�Y
> Recv: Y\x10YbYmYwY|Y�Y�Y����������\x0e�����������������\x0e�\x01�\x05�	�v�{�������
> �����������������
> �����\x03�� = 0x80:09:00:00
> 
>

livid-fpv, 2018-03-08:
Baudrate?

8Roman8, 2018-03-08:
no, the baudrate is right, i know it looks so ^^

all other commands are readable, printing over usb is possible too.

teemuatlut, 2018-03-08:
I presume you're using a very recent commit version of the branch? There was a recent change to move stepper names into PROGMEM and that might have broken some things.
I saw this yesterday as well but with Printrun it wasn't that terrible.

8Roman8, 2018-03-08:
Yes, i use the newest branch...

@teemuatlut
I had an another issue few days ago, 
I activated sensorless homing, x and y works well, but only one z axis detect a stall... (dual z motor with dual z-endstop). 
Eventuell with sensivity 1.

teemuatlut, 2018-03-08:
The Z axis sensorless homing is meant for delta printers only and I really don't like that it was added like that. Someone at some point will break their machine with it.

8Roman8, 2018-03-08:
Ok I understand that, maybe this can be implemented only in the case when the machine homes to z-max at cartesian printers? I don't know all the different machine types, but I think at z-max the most machines should be safe.

thinkyhead, 2018-03-09:
Aha, progmem strings in an array of progmem pointers needs one extra step! This should now be patched.

8Roman8, 2018-03-09:
Perfect!

> Send: M122
Recv: 		X	Y	Z	Z2	E0	E1
Recv: Enabled		false	false	false	false	false	false
Recv: Set current	950	950	600	600	600	600
Recv: RMS current	939	939	581	581	581	581
Recv: MAX current	1324	1324	819	819	819	819
Recv: Run current	16/31	16/31	18/31	18/31	18/31	18/31
Recv: Hold current	8/31	8/31	9/31	9/31	9/31	9/31
Recv: CS actual		8/31	8/31	9/31	9/31	9/31	9/31
Recv: PWM scale	0	0	0	0	0	0
Recv: vsense		0=.325	0=.325	1=.18	1=.18	1=.18	1=.18
Recv: stealthChop	true	true	true	true	true	true
Recv: msteps		16	16	16	16	16	16
Recv: tstep		1048575	1048575	1048575	1048575	1048575	1048575
Recv: pwm
Recv: threshold		79	79	82	82	164	168
Recv: [mm/s]		100.08	100.08	3.01	3.01	30.02	30.07
Recv: OT prewarn	false	false	false	false	false	false
Recv: OT prewarn has
Recv: been triggered	false	false	false	false	false	false
Recv: off time		5	5	5	5	5	5
Recv: blank time	24	24	24	24	24	24
Recv: hysterisis
Recv: -end		2	2	2	2	2	2
Recv: -start		3	3	3	3	3	3
Recv: Stallguard thrs	2	2	8	8	0	0
Recv: DRVSTATUS	X	Y	Z	Z2	E0	E1
Recv: stallguard
Recv: sg_result		0	0	0	0	0	0
Recv: fsactive
Recv: stst		X	X	X	X	X	X
Recv: olb
Recv: ola
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x80:08:00:00
Recv: 	Y = 0x80:08:00:00
Recv: 	Z = 0x80:09:00:00
Recv: 	Z2 = 0x80:09:00:00
Recv: 	E0 = 0x80:09:00:00
Recv: 	E1 = 0x80:09:00:00

commit 03676d8e (2018-03-09): Fix `_tmc_say_axis` string address

Fixes #10003
commit cc59ac46 (2018-03-09): Fix _tmc_say_axis string address

Fixes #10003
closed 2018-03-09 by 8Roman8