MarlinFirmware/Marlin #10005: [bugfix- 2.0.x]   arduino due+RAMPS_FD_V2 compile error

cjsoong, 2018-03-08:
bugfix 2.0.x  arduino due+RAMPS_FD_V2 compile error
problem causes :  Make enums into implicit char 
[2.0.x] Move tmc_util.* extended_axis_codes to PROGMEM #9974
ERROR MSG:
Arduino: 1.8.5 (Windows 10), Board: "Arduino Due (Programming Port)"

In file included from sketch\src\HAL\HAL_DUE\usb\sysclk.h:51:0,

                 from sketch\src\HAL\HAL_DUE\usb\sysclk.c:49:

sketch\src\HAL\HAL_DUE\usb\pll.h:79:17: error: expected identifier or '(' before ':' token

 enum pll_source : char {

                 ^

sketch\src\HAL\HAL_DUE\usb\pll.h:116:19: error: parameter 2 ('e_src') has incomplete type

   enum pll_source e_src, uint32_t ul_div, uint32_t ul_mul)

                   ^

sketch\src\HAL\HAL_DUE\usb\pll.h:205:54: error: parameter 1 ('e_src') has incomplete type

 static inline void pll_enable_source(enum pll_source e_src)

                                                      ^

sketch\src\HAL\HAL_DUE\usb\pll.h: In function 'pll_enable_source':

sketch\src\HAL\HAL_DUE\usb\pll.h:208:7: error: 'PLL_SRC_MAINCK_4M_RC' undeclared (first use in this function)

  case PLL_SRC_MAINCK_4M_RC:

       ^

sketch\src\HAL\HAL_DUE\usb\pll.h:208:7: note: each undeclared identifier is reported only once for each function it appears in

sketch\src\HAL\HAL_DUE\usb\pll.h:209:7: error: 'PLL_SRC_MAINCK_8M_RC' undeclared (first use in this function)

  case PLL_SRC_MAINCK_8M_RC:

       ^

sketch\src\HAL\HAL_DUE\usb\pll.h:210:7: error: 'PLL_SRC_MAINCK_12M_RC' undeclared (first use in this function)

  case PLL_SRC_MAINCK_12M_RC:

       ^

sketch\src\HAL\HAL_DUE\usb\pll.h:211:7: error: 'PLL_SRC_MAINCK_XTAL' undeclared (first use in this function)

  case PLL_SRC_MAINCK_XTAL:

       ^

sketch\src\HAL\HAL_DUE\usb\pll.h:212:7: error: 'PLL_SRC_MAINCK_BYPASS' undeclared (first use in this function)

  case PLL_SRC_MAINCK_BYPASS:

       ^

In file included from sketch\src\HAL\HAL_DUE\usb\compiler.h:53:0,

                 from sketch\src\HAL\HAL_DUE\usb\osc.h:50,

                 from sketch\src\HAL\HAL_DUE\usb\sysclk.h:50,

                 from sketch\src\HAL\HAL_DUE\usb\sysclk.c:49:

sketch\src\HAL\HAL_DUE\usb\pll.h: In function 'pll_enable_config_defaults':

sketch\src\HAL\HAL_DUE\usb\conf_clock.h:74:37: error: 'PLL_SRC_MAINCK_XTAL' undeclared (first use in this function)

 #define CONFIG_PLL0_SOURCE          PLL_SRC_MAINCK_XTAL

                                     ^

sketch\src\HAL\HAL_DUE\usb\pll.h:233:21: note: in expansion of macro 'CONFIG_PLL0_SOURCE'

   pll_enable_source(CONFIG_PLL0_SOURCE);

                     ^

sketch\src\HAL\HAL_DUE\usb\sysclk.c: In function 'sysclk_enable_usb':

sketch\src\HAL\HAL_DUE\usb\conf_clock.h:74:37: error: 'PLL_SRC_MAINCK_XTAL' undeclared (first use in this function)

 #define CONFIG_PLL0_SOURCE          PLL_SRC_MAINCK_XTAL

                                     ^

sketch\src\HAL\HAL_DUE\usb\sysclk.c:82:21: note: in expansion of macro 'CONFIG_PLL0_SOURCE'

   pll_enable_source(CONFIG_PLL0_SOURCE);

                     ^

exit status 1
Error compiling for board Arduino Due (Programming Port).

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Bob-the-Kuhn, 2018-03-08:
Confirmed - just compiling for Due gives the above errors.

If I compile for LPC1768 with no issues.

If I back out the ENUM changes in the \Marlin\src\HAL\HAL_DUE\usb directory then it compiles.

I took some gueses at at fixes with no luck.  I tried converting the tabs to spaces, different EOL settings, retype lines to see if there were unprintable characters and all kinds of includes.

Here's the files with the enum mods backed out in case you want to get going before this is fixed.

[enum mod .zip](https://github.com/MarlinFirmware/Marlin/files/1794525/enum.mod.zip)

Bob-the-Kuhn, 2018-03-09:
PR #10017 has been created as a temporary work around

Bob-the-Kuhn, 2018-03-09:
The PR has been merged so you can compile DUE again.

ejtagle, 2018-03-24:
this was fixed by PR #10017

closed 2018-03-24 by cjsoong