MarlinFirmware/Marlin #10006: [FR] Heating speed, heating too fast breaks glass

ArtemKuchin, 2018-03-08:
Usually  people have glass/mirror on top of aluminum bed or just on top of heater.  In both cases glass is just clamped to bed (heater) which are screwed to the carriage.
In my case i have  a fully glass (mirror)  bed with glued screws and screwed to the carriage. Then heater is glued to the glass: https://www.thingiverse.com/thing:2805123

I've seen reports of cracked glass from many users who heat bed over 70C.

Before today i printed with bed temperature up to 75C and never had a problem. But today, i installed some thermal insulation under table (never needed it before) and just pushed heat to 80C. Waited about 5 minutes and got a crack. Terrible.

After rereading reports from others and thinking about what happened to my bed it seems like the reason is in the heating speed, not the temperature itself. I installed the thermal insulation  and that increased the heating speed about 2x and temperature gradient in the mirror became too large, expansion too uneven and it cracked. Seem like it is a common reason for the cracked glass bed.

So, the solution is very simple: slow down the heating. I guess it can be done manually or using some magic scripting in the slicer, but would be better to have maximum heating speed (C/min) while heating in the marlin config.

Maybe it can already be done by tweaking PID settings? I just don't know.

Sebastianv650, 2018-03-08:
This function is available in Confguation.h, search for #define MAX_BED_POWER.

ArtemKuchin, 2018-03-08:
Yes, i know this param. However, as i inderstood, setting it to, for example, 127, does not reduce heating speed 2 or 3 times (it is not linear system), but just does what it says  - limits the powers. So, eventually, it will limit hot i can set the bed and that;s not what i want. When the bed is already heated to the target temperature not problem at max power. Reducing max power will probably make harder to keep set temperature for PID. Also, setting it lower does not actually help reduce speed of heating in the beginning (at lower temperatures) much.

Sebastianv650, 2018-03-08:
Are you using tempered glass for the build plate? In fact I never heared that before, in most cases I see people increasing the heater wattage up into the kW range for faster heating.

Studiodyne, 2018-03-08:
I have a 400x400 3mm glass with direct heating pad glued
My heatpad is 600watts , and i use it at full power
'Vitroceramic or Borosilicate ' are the glass where i make my cook on my kitchen and this type of glass never breaks  with high power and high speed heating

Change your glass , it's not a good quality
To have low price , you can buy on fireplace store , or in DIY store

Sineos, 2018-03-08:
> In my case i have a fully glass (mirror) bed with glued screws and screwed to the carriage. Then heater is glued to the glass: https://www.thingiverse.com/thing:2805123

I don't think it is because of cheap glass. It is because of the way it is mounted. The glass cannot compensate its own thermal expansion (screws) and it cannot compensate the thermal expansion of the rigid heat pcb (glued). At least not fast enough to avoid too much tension causing it to break.

I guess @ArtemKuchin analysis is correct that a reduced heat up speed will solve the issue.

ArtemKuchin, 2018-03-08:
@Sineos the glue for screws is not 100% rigid, i can move angle the screws a little with hand, the heater is mounted using pure silicon in point pattern (applied the glue i points with thermal compound in all other areas). So, the glues should not hold the glass too much. But this is just a guess. Anyway, before i added thermal insulation i tested this setup up to 85C several times. Take long time to heat up but no problems.

@Studiodyne  I use usual plan 3mm mirror. Cheap, but very very flat. Also mirror layer help redistributing the heat. Borosilicate of course is better, but my size is non standard. 23x23cm, hard to find. I am not sure how to cut borosilicate at home.

@Sebastianv650 my guess that KW heaters are used for aluminum plate heater with borosilicate on top.

Studiodyne, 2018-03-09:
@ArtemKuchin 
Fireplace store cut the glass whatever the dimensions ,   DIY store or tools store cut the glass at all dimensions , the dimensions is not a difficulty

I love this topic , because i have a problem , i have the pad glued on a transparent glass , and it's not really ' beautiful' , i need idea to have a good ' finish '
The mirror is really a good idea , but , the dirty surface is very visible , more than my own ben 

Do you have idea to have a beautifull black surface ,  ' no buildtak because SHIT ! '

closed 2018-03-11 by thinkyhead