MarlinFirmware/Marlin #10007: [bugfix_1.1.x] M600 (ADVANCED_PAUSE_FEATURE) not working correctly

TimotF, 2018-03-08:
### Description

I just added a new filament runout sensor wich uses the M600 routine. The problem appears when I run teh M600 command : the filament won't unload or load (retract and purge working).

### Steps to Reproduce

1. enable "ADVANCED_PAUSE_FEATURE"
2. run M600

**Expected behavior:** 

- fisrt retract
- park nozzle
- unload filament
- wait for the user to press the button
- load filament
- purge
- resume

**Actual behavior:** 

- first retract (PAUSE_PARK_RETRACT_LENGTH)  => working
- park nozzle => working
- unload (FILAMENT_CHANGE_UNLOAD) => **Not working**
- wait until we press the button => working
- load (FILAMENT_CHANGE_LOAD) => **Not working**
- purge (ADVANCED_PAUSE_EXTRUDE) => working

#### Additional Information

conf used :
```
#define ADVANCED_PAUSE_FEATURE
#if ENABLED(ADVANCED_PAUSE_FEATURE)
  #define PAUSE_PARK_RETRACT_FEEDRATE 60      // (mm/s) Initial retract feedrate.
  #define PAUSE_PARK_RETRACT_LENGTH 2         // (mm) Initial retract.
                                              // This short retract is done immediately, before parking the nozzle.
  #define FILAMENT_CHANGE_UNLOAD_FEEDRATE 20  // (mm/s) Unload filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_UNLOAD_LENGTH 850   // (mm) The length of filament for a complete unload.
                                              //   For Bowden, the full length of the tube and nozzle.
                                              //   For direct drive, the full length of the nozzle.
                                              //   Set to 0 for manual unloading.
  #define FILAMENT_CHANGE_LOAD_FEEDRATE 5     // (mm/s) Load filament feedrate. This can be pretty fast.
  #define FILAMENT_CHANGE_LOAD_LENGTH 815       // (mm) Load length of filament, from extruder gear to nozzle.
                                              //   For Bowden, the full length of the tube and nozzle.
                                              //   For direct drive, the full length of the nozzle.
  #define ADVANCED_PAUSE_EXTRUDE_FEEDRATE 3   // (mm/s) Extrude feedrate (after loading). Should be slower than load feedrate.
  #define ADVANCED_PAUSE_EXTRUDE_LENGTH 30    // (mm) Length to extrude after loading.
                                              //   Set to 0 for manual extrusion.
                                              //   Filament can be extruded repeatedly from the Filament Change menu
                                              //   until extrusion is consistent, and to purge old filament.

                                              // Filament Unload does a Retract, Delay, and Purge first:
  #define FILAMENT_UNLOAD_RETRACT_LENGTH 13   // (mm) Unload initial retract length.
  #define FILAMENT_UNLOAD_DELAY 5000          // (ms) Delay for the filament to cool after retract.
  #define FILAMENT_UNLOAD_PURGE_LENGTH 8      // (mm) An unretract is done, then this length is purged.

  #define PAUSE_PARK_NOZZLE_TIMEOUT 450        // (seconds) Time limit before the nozzle is turned off for safety.
  #define FILAMENT_CHANGE_ALERT_BEEPS 10      // Number of alert beeps to play when a response is needed.
  #define PAUSE_PARK_NO_STEPPER_TIMEOUT       // Enable for XYZ steppers to stay powered on during filament change.

  #define PARK_HEAD_ON_PAUSE                // Park the nozzle during pause and filament change.
  //#define HOME_BEFORE_FILAMENT_CHANGE       // Ensure homing has been completed prior to parking for filament change

  #define FILAMENT_LOAD_UNLOAD_GCODES       // Add M701/M702 Load/Unload G-codes, plus Load/Unload in the LCD Prepare menu.
  //#define FILAMENT_UNLOAD_ALL_EXTRUDERS     // Allow M702 to unload all extruders above a minimum target temp (as set by M302)
#endif
```

I first thought it was the typical "EXTRUDE_MAXLENGTH" problem, but then I checked and it was already set :
```
#define PREVENT_LENGTHY_EXTRUDE
#define EXTRUDE_MAXLENGTH 2000
```
I downloaded the latest bugfix_1.1.x and I had the same behaviour...

[config.zip](https://github.com/MarlinFirmware/Marlin/files/1794144/config.zip)

AnHardt, 2018-03-08:
```
// This option prevents a single extrusion longer than EXTRUDE_MAXLENGTH.
// Note that for Bowden Extruders a too-small value here may prevent loading.
#define PREVENT_LENGTHY_EXTRUDE
#define EXTRUDE_MAXLENGTH 200
```
???

Studiodyne, 2018-03-08:
Please , can you make a 2.0 release , because , i use it and i'm sure of this release
The bugfix 1.1.x haven't all updates or sometimes , it's 2.0 that is not updated
But , for many reason it's more easy to work on 2.0 because , all is separate , all modules , functions have their own units

But , for you , and because , I use pause everytime , i can confirm all is ok , on 2.0
Put your 2 cfg files and try to compil 
If not compile , search in the issue tab , all is explained of how to compile a 2.0 ' very easy' 

and make your tests , and give feedback 

steeve

@Thinkyhead   , i've seen the ' parser.codenum ' not implemented in bugfix 1.1.x ... ? Why or soon .. I know i'm not the perfect guy to suggest something here but .... please , stop 1.1.x  and leave just ONE RELEASE 2.0 & bugfix 2.0 .... too much wasted time for you , and so many differences to fix eternally 
I'm sure you agree with all i say

TimotF, 2018-03-09:
Thanks for your answers. I updated the firmware to V2.0 and it indeed works. What I don't understand is when I changed the firmware back to the V1.1.x, it also worked ! After checking the code, my guess is I had for some reason a problem when resetting the EEPROM config and as `FILAMENT_CHANGE_UNLOAD_LENGTH` and `FILAMENT_CHANGE_LOAD_LENGTH` are stored in EEPROM, it was indeed 0 instead of the value I wanted...

My apologies for the inconvenience.

closed 2018-03-09 by TimotF

Studiodyne, 2018-03-09:
Delete  temp folder 
May be a confusion with old builts

macropin, 2019-09-02:
@TimotF Thanks for detailing this. I hit the same issue.

Running "Initialise EEPROM" from the menu seemed to do the trick!