MarlinFirmware/Marlin #10009: Watchdog reset on marlin boot while loading UBL mesh

LichtiMC, 2018-03-08:
`Recv: start
Recv: echo:PowerUp
Recv: Marlin bugfix-2.0.x
Recv: 
Recv: echo: Last Updated: 2018-01-20 | Author: David
Send: N0 M110 N0*125
Recv: echo:Compiled: Mar  8 2018
Recv: echo: Free Memory: 83808  PlannerBufferBytes: 1344
Recv: Unified Bed Leveling System v1.01 inactive.
Recv: echo:V52 stored settings retrieved (617 bytes; crc 15378)
Recv: Unified Bed Leveling System v1.01 inactive.
Recv: 
Recv: Unified Bed Leveling initialized.
Recv: 
Recv: Mesh loaded from slot 0
Recv: Mesh 0 loaded from storage.
Recv: echo:  G21    ; Units in mm
Recv: echo:  M149 C ; Units in Celsius
Recv: 
Recv: echo:Filament settings: Disabled
Recv: echo:  M200 D1.75
Recv: echo:  M200 D0
Recv: echo:Steps per unit:
Recv: echo:  M92 X79.90 Y79.82 Z400.00 E105.00
Recv: echo:Maximum feedrates (units/s):
Recv: echo:  M203 X300.00 Y300.00 Z5.00 E25.00
Recv: echo:Maximum Acceleration (units/s2):
Recv: echo:  M201 X1000 Y1000 Z100 E10000
Recv: echo:Acceleration (units/s2): P<print_accel> R<retract_accel> T<travel_accel>
Recv: echo:  M204 P1000.00 R1000.00 T1000.00
Recv: echo:Advanced: S<min_feedrate> T<min_travel_feedrate> B<min_segment_time_us> X<max_xy_jerk> Z<max_z_jerk> E<max_e_jerk>
Recv: echo:  M205 S0.00 T0.00 B20000 X10.00 Y10.00 Z0.30 E5.00
Recv: echo:Home offset:
Recv: echo:  M206 X0.00 Y0.00 Z0.00
Recv: echo:Unified Bed Leveling:
Recv: echo:  M420 S0 Z10.00
Recv: 
Recv: Unified Bed Leveling System v1.01 inactive.
Recv: 
Recv: Active Mesh Slot: 0
Recv: EEPROM can hold 8 meshes.
Recv: 
Recv: echo:  G29 I 999
Recv: echo:  M421 I0 J0 Z-0.462500 ; X1.00, Y1.00
Recv: echo:  M421 I0 J1 Z-0.487500 ; X1.00, Y25.22
Recv: echo:  M421 I0 J2 Z-0.432500 ; X1.00, Y49.44
Recv: echo:  M421 I0 J3 Z-0.252500 ; X1.00, Y73.67
Recv: echo:  M421 I0 J4 Z-0.317500 ; X1.00, Y97.89
Recv: echo:  M421 I0 J5 Z-0.252500 ; X1.00, Y122.11
Recv: echo:  M421 I0 J6 Z-0.200000 ; X1.00, Y146.33
Recv: echo:  M421 I0 J7 Z-0.025000 ; X1.00, Y170.56
Recv: echo:  M421 I0 J8 Z-0.042500 ; X1.00, Y194.78
Recv: echo:  M421 I0 J9 Z0.245000 ; X1.00, Y219.00
Recv: echo:  M421 I1 J0 Z-0.462500 ; X25.22, Y1.00
Recv: echo:  M421 I1 J1 Z-0.487500 ; X25.22, Y25.22
Recv: echo:  M421 I1 J2 Z-0.432500 ; X25.22, Y49.44
Recv: echo:  M421 I1 J3 Z-0.262500 ; X25.22, Y73.67
Recv: echo:  M421 I1 J4 Z-0.317500 ; X25.22, Y97.89
Recv: echo:  M421 I1 J5 Z-0.252500 ; X25.22, Y122.11
Recv: echo:  M421 I1 J6 Z-0.200000 ; X25.22, Y146.33
Recv: echo:  M421 I1 J7 Z-0.025000 ; X25.22, Y170.56
Recv: echo:  M421 I1 J8 Z-0.042500 ; X25.22, Y194.78
Recv: echo:  M421 I1 J9 Z0.177500 ; X25.22, Y219.00
Recv: echo:  M421 I2 J0 Z-0.462500 ; X49.44, Y1.00
Recv: echo:  M421 I2 J1 Z-0.487500 ; X49.44, Y25.22
Recv: echo:  M421 I2 J2 Z-0.432500 ; X49.44, Y49.44
Recv: echo:  M421 I2 J3 Z-0.272500 ; X49.44, Y73.67
Recv: echo:  M421 I2 J4 Z-0.317500 ; X49.44, Y97.89
Recv: echo:  M421 I2 J5 Z-0.252500 ; X49.44, Y122.11
Recv: echo:  M421 I2 J6 Z-0.200000 ; X49.44, Y146.33
Recv: echo:  M421 I2 J7 Z-0.025000 ; X49.44, Y170.56
Recv: echo:  M421 I2 J8 Z-0.042500 ; X49.44, Y194.78
Recv: echo:  M421 I2 J9 Z0.110000 ; X49.44, Y219.00
Recv: echo:  M421 I3 J0 Z-0.335000 ; X73.67, Y1.00
Recv: echo:  M421 I3 J1 Z-0.385000 ; X73.67, Y25.22
Recv: echo:  M421 I3 J2 Z-0.222500 ; X73.67, Y49.44
Recv: echo:  M421 I3 J3 Z-0.282500 ; X73.67, Y73.67
Recv: echo:  M421 I3 J4 Z-0.192500 ; X73.67, Y97.89
Recv: echo:  M421 I3 J5 Z-0.210000 ; X73.67, Y122.11
Recv: echo:  M421 I3 J6 Z-0.057500 ; X73.67, Y146.33
Recv: echo:  M421 I3 J7 Z0.005000 ; X73.67, Y170.56
Recv: echo:  M421 I3 J8 Z0.007500 ; X73.67, Y194.78
Recv: echo:  M421 I3 J9 Z0.042500 ; X73.67, Y219.00
Recv: echo:  M421 I4 J0 Z-0.230000 ; X97.89, Y1.00
Recv: echo:  M421 I4 J1 Z-0.342500 ; X97.89, Y25.22
Recv: echo:  M421 I4 J2 Z-0.302500 ; X97.89, Y49.44
Recv: echo:  M421 I4 J3 Z-0.275000 ; X97.89, Y73.67
Recv: echo:  M421 I4 J4 Z-0.100000 ; X97.89, Y97.89
Recv: echo:  M421 I4 J5 Z-0.132500 ; X97.89, Y122.11
Recv: echo:  M421 I4 J6 Z-0.040000 ; X97.89, Y146.33
Recv: echo:  M421 I4 J7 Z-0.167500 ; X97.89, Y170.56
Recv: echo:  M421 I4 J8 Z-0.002500 ; X97.89, Y194.78
Recv: echo:  M421 I4 J9 Z0.142500 ; X97.89, Y219.00
Recv: echo:  M421 I5 J0 Z-0.210000 ; X122.11, Y1.00
Recv: echo:  M421 I5 J1 Z-0.112500 ; X122.11, Y25.22
Recv: echo:  M421 I5 J2 Z-0.175000 ; X122.11, Y49.44
Recv: echo:  M421 I5 J3 Z-0.207500 ; X122.11, Y73.67
Recv: echo:  M421 I5 J4 Z-0.142500 ; X122.11, Y97.89
Recv: echo:  M421 I5 J5 Z-0.197500 ; X122.11, Y122.11
Recv: echo:  M421 I5 J6 Z-0.170000 ; X122.11, Y146.33
Recv: echo:  M421 I5 J7 Z-0.152500 ; X122.11, Y170.56
Recv: start
Recv: echo: Watchdog Reset`

teemuatlut, 2018-03-08:
I've had this too.
If you want to fix this yourself, change the `safe_delay(75)` to `safe_delay(25)` in [ubl.cpp](https://github.com/MarlinFirmware/Marlin/blob/bugfix-2.0.x/Marlin/src/feature/bedlevel/ubl/ubl.cpp#L66).

LichtiMC, 2018-03-08:
Working... thank you!

viper93458, 2018-03-09:
@teemuatlut thank you for the reference. I will attempt this fix however I don’t see a watchdog reset message. Also, I am using a Mega2560. I will report back tomorrow.

thinkyhead, 2018-03-09:
This has now been patched a little more definitively by moving the watchdog init to the end of `setup()`. The 75ms safe delay can remain, or it can be reduced. At your discretion.

closed 2018-03-09 by thinkyhead