MarlinFirmware/Marlin #10010: Move G42 to G44 to allow for CNC G42/G43

thinkyhead, 2018-03-08:
The `G42` command is currently assigned to "Move to Mesh Point." As this is a less essential feature and not used inside of sliced G-code, I propose to move this function to the unassigned G-code (both in RepRap and in CNC) `G44` so that we can make room for the `G42` and `G43` commands which are used in CNC for Dynamic Cutter Compensation (radius and left/right of path) and for Tool Length Offset.

GMagician, 2018-03-08:
CNC machine uses G40, G41 and G42 for tool compensation, G43 is not used for that purpose as I know

thinkyhead, 2018-03-09:
As I'm implementing CNC features for a client, in the process I'm trying to maintain compatible codes (following LinuxCNC guidance), and also aiming to retain the expected behavior of CNC G-code — such as permitting multiple `Gnn` codes within a single line of G-code. While I cannot merge these features for a while, it seems like a good idea to clear out space early.

GMagician, 2018-03-09:
There is also the possibility to have same gcode that act in a different way depending on "machine typology" this, maybe, can be a solution for historical commands that cannot be moved.

thinkyhead, 2018-03-09:
True. I think it's relatively safe to deprecate `M42` in favor of `M44`. Most of the time when codes are added we are more careful to check both RepRap.org and LinuxCNC.org.

TheSFReader, 2018-03-09:
M42 in favor of M44 or G42 in favor of G44 ?

boelle, 2019-10-24:
@thinkyhead  did you do the move of G42 to G44 ?

boelle, 2019-11-24:
this one has gone stale and the discussion could be done on discord as well 

Marlin Discord server. Join link: https://discord.gg/n5NJ59y

closed 2019-11-24 by boelle