MarlinFirmware/Marlin #10011: [bugfix-2.0.x] RAMPS_FD_V2.X Pins wrong

dembach, 2018-03-08:
Please add following lines to pins_RAMPS_FD_V2.h:
```
#ifndef PS_ON_PIN
  #define PS_ON_PIN        12
#endif

#ifndef FILWIDTH_PIN
  #define FILWIDTH_PIN      5   // Analog Input on AUX2
#endif
```
Both possibilities are missing.

Additionally there are 2 Serial Ports on RAMPS_FD, so why there is no definition for this ?

Please add following lines to pins_RAMPS_FD.h:
```
#if ENABLED(HAVE_TMC2208)
  /**
   * TMC2208 stepper drivers
   *
   * Hardware serial communication ports.
   * If undefined software serial is used according to the pins below
   */
  //#define X_HARDWARE_SERIAL  Serial1
  //#define X2_HARDWARE_SERIAL Serial1
  //#define Y_HARDWARE_SERIAL  Serial1
  //#define Y2_HARDWARE_SERIAL Serial1
  //#define Z_HARDWARE_SERIAL  Serial1
  //#define Z2_HARDWARE_SERIAL Serial1
  //#define E0_HARDWARE_SERIAL Serial1
  //#define E1_HARDWARE_SERIAL Serial1
  //#define E2_HARDWARE_SERIAL Serial1
  //#define E3_HARDWARE_SERIAL Serial1
  //#define E4_HARDWARE_SERIAL Serial1
#endif
```
Also Support for Spindle / Laser is missing.
```
//
// M3/M4/M5 - Spindle/Laser Control
//
#if ENABLED(SPINDLE_LASER_ENABLE) && !PIN_EXISTS(SPINDLE_LASER_ENABLE)
  #if HOTENDS < 3
    #define SPINDLE_LASER_ENABLE_PIN  45  // Use E2 ENA
    #define SPINDLE_LASER_PWM_PIN     12  // MUST BE HARDWARE PWM
    #define SPINDLE_DIR_PIN           47 // Use E2 DIR
  #endif
#endif
```

[RAMPS-FD-Schematic.pdf](https://github.com/MarlinFirmware/Marlin/files/1794861/RAMPS-FD-Schematic.pdf)

maybe @bobc could complete the listing, what else is missing , especially in V2.x

thanks

Bob-the-Kuhn, 2018-03-10:
I'll include these changes in PR #9961.

Bob-the-Kuhn, 2018-03-10:
Changes have been merged.

closed 2018-03-10 by Bob-the-Kuhn