MarlinFirmware/Marlin #9990: Object Sheared while Printing

DBSETIAWAN, 2018-03-08:
Hi !
I faced a problem while printing. 

First, my axes move inaccurately while prepare. And the problem was finish using this thread:
https://github.com/MarlinFirmware/Marlin/issues/9955

But after XYZ-Axes move correctly, I faced another problem. While printing, the object is shifting in the second layaer and go on into the next layer.

I used Marlin 1.0.x, and the object that I was printing is Calibration Cube:
https://intip.in/8F5Q

And the object that I print:
https://intip.in/Pict1
https://intip.in/Pict2

Before this, I was try 139 objects that all was shifting and broken while printing, and I tried to find the problems in the code and hardware, but finally Im surrender. Please help me out to this problem.

thinkyhead, 2018-03-08:
Make sure your acceleration and jerk settings aren't too high.

---
This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. For best results getting help with configuration and troubleshooting, please use the following community resources:
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/).
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)

DBSETIAWAN, 2018-03-08:
In my Marlin, on my Config*.h, there are settings for:

#define DEFAULT_AXIS_STEPS_PER_UNIT   { 80, 80, 412.9032258064, 200  }
#define DEFAULT_MAX_FEEDRATE             { 400, 400, 2, 90 }
#define DEFAULT_MAX_ACCELERATION     { 5000, 5000, 50, 5000 }

#define DEFAULT_XJERK                 20.0
#define DEFAULT_YJERK                 20.0
#define DEFAULT_ZJERK                  0.4
#define DEFAULT_EJERK                  5.0

Is it the part of codes that you mention @thinkyhead ?
Thank you for your fast response.

thinkyhead, 2018-03-08:
You can tune these values with G-code. See:
- http://marlinfw.org/docs/gcode/M201.html
- http://marlinfw.org/docs/gcode/M203.html
- http://marlinfw.org/docs/gcode/M204.html

After tuning you can save with `M500`. (See http://marlinfw.org/docs/gcode/M500.html)

DBSETIAWAN, 2018-03-08:
Actually, I don't understand with the explanation on the Marlin's site.
Please anyone can help me to make me more understand about that order.
Thanks a lot.

th33xitus, 2018-03-12:
You have to lower your acceleration settings.
Accelerations of 5000mm/s² seem to be too high for your setup so your printer is loosing steps -> layer shifting occurs. Google for "layer shifting" and you will find several explanations and ways so solve the issue.

This has nothing to do with the firmware itself, its rather just a bad configuration and not really belongs into here.

nechveirst, 2018-03-26:
Im so sorry for late reply this post.
Thanks @th33xitus @thinkyhead , after I've tried thigten belt, setting the belt and current of motor driver, and applied some combination of reducing a few acceleration and made improvements in Config*.h

`#define DEFAULT_ACCELERATION          1000   
#define DEFAULT_RETRACT_ACCELERATION  1000   
#define DEFAULT_TRAVEL_ACCELERATION    500 

#define DEFAULT_XJERK                 20.0
#define DEFAULT_YJERK                 20.0
#define DEFAULT_ZJERK                  0.2
#define DEFAULT_EJERK                  5.0`

and the result was so fascinating. The surfaces and infill patern are so smooth and nice. I've printed calibrated cube 2x2 cm for three times.

[image]
[image]
[image]

But there is a new problem that I faced. As you can see in my pictures, the 3-last layer in the top is shifting. Are there any settings that should I changed for solve that problem ?
Thank you so much.

thinkyhead, 2018-03-28:
Your belt might now be **too** tight. Layer shifting as a result of bad tuning or physical printer setup is not a bug in Marlin, so you shouldn't be seeking help here where we focus on development and solving bugs. For best results getting help with configuration and troubleshooting, please use the following resources:
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)

closed 2018-03-28 by thinkyhead