MarlinFirmware/Marlin #10022: MKS BASE 1.4 Stepper Pins

rt1970, 2018-03-09:
I have MKS BASE 1.4 Board with HR4982 Stepper.
I uncomment in Pins_MKS_BASE.h the PIN-Config for MS1 and MS2

Here is the error (and many more lines):
In file included from sketch\src\module\../inc/../HAL/HAL_AVR/HAL_AVR.h:46:0,
                 from sketch\src\module\../inc/../HAL/HAL.h:38,
                 from sketch\src\module\../inc/MarlinConfig.h:28,
                 from sketch\src\module\stepper_indirection.h:47,
                 from sketch\src\module\stepper.h:46,
                 from sketch\src\module\stepper.cpp:47:
sketch\src\module\../inc/../pins/pins_MKS_BASE.h:42:33: error: pasting "/* Digital 3  / Pin 5   / PE3*/" and "_DDR" does not give a valid preprocessing token
#define X_MS1_PIN           5   // Digital 3  / Pin 5   / PE3
                                ^
sketch\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:90:36: note: in definition of macro '_SET_OUTPUT'
 #define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)
                                   ^
sketch\src\module\stepper.cpp:1514:5: note: in expansion of macro 'SET_OUTPUT'
    SET_OUTPUT(X_MS1_PIN);
    ^
sketch\src\module\stepper.cpp:1514:16: note: in expansion of macro 'X_MS1_PIN'
    SET_OUTPUT(X_MS1_PIN);
               ^
sketch\src\module\../inc/../pins/pins_MKS_BASE.h:42:33: error: pasting "/* Digital 3  / Pin 5   / PE3*/" and "_PIN" does not give a valid preprocessing token
#define X_MS1_PIN           5   // Digital 3  / Pin 5   / PE3
                                ^
sketch\src\module\../inc/../HAL/HAL_AVR/fastio_AVR.h:90:61: note: in definition of macro '_SET_OUTPUT'
#define _SET_OUTPUT(IO) do {DIO ## IO ## _DDR |= _BV(DIO ## IO ## _PIN); } while (0)
                                                         ^
sketch\src\module\stepper.cpp:1514:5: note: in expansion of macro 'SET_OUTPUT'
     SET_OUTPUT(X_MS1_PIN);
     ^

8Roman8, 2018-03-09:
Which arduino ide du you use? 

I had the same issue, I had to delete the comments on the lines with "/*" in the error code.. 

My English is very bad but I hope you understand me^^

rt1970, 2018-03-09:
I use Arduino 1.9.0 beta
I have another try with Marlin 2.x It is new Board MKS_BASE_HEROIC. Thats my stepper-ICs. But same errors. Sorry for my bad english. Im from germany.

8Roman8, 2018-03-09:
Probier mal die Kommentare hinter den Definitionen zu löschen in den Zeilen die in der Fehlermeldung auftauchen...

`#define X_MS1_PIN 5 // Digital 3 / Pin 5 / PE3` --> `#define X_MS1_PIN 5`

rt1970, 2018-03-09:
Tatsächlich! Wär ich nie drauf gekommen! // ist eigentlich immer Kommentar... Geht wohl nur in neuer Zeile beim Arduino... VIELEN Dank! Mal sehen ob ihm die Leerzeichen vor den PIN-Ziffern auch stören oder ob ich jetzt (endlich) 128 Step´s einstellen kann. Ist schon ziemlich laut der Tevo Tornado...

8Roman8, 2018-03-09:
Da bin ich überfragt, ich hab die Treiber bis gerade eben gar nicht gekannt sry..

rt1970, 2018-03-09:
? Die Rechnung kommt mir komisch vor... Kommt immer 128 raus...
Hab jetzt an der T8 Welle Z=400 Steps/mm. Sollte also auf 1/4 Microschritte laufen. Wieviel Steps sind es dann bei 1/128?

8Roman8, 2018-03-09:
Genau

rt1970, 2018-03-09:
So. Getestet. Firmware Marlin 2.0 bugfix
M350 X128 zeigt mir auch X: 01 an. Aber die X-Achse ruckelt nur ganz langsam zum Zielpunkt. Auch beim X-Homing ruckelt und knattert er nur... Mit 2570 fährt er zu weit. geschätzt will er um die 1200 haben. Schon alles merkwürdig. Schade!

8Roman8, 2018-03-09:
ich hab gerade gesehen, dass das board einen 8 bit microcontroller hat, das ist für 128 microsteps einfach zu langsam... Ich bin davor ausgegangen dass es ein 32bit contoller ist...

ich würde höchstens auf 32 microsteps gehen sonst hast du immer Probleme, dass der controller nicht mitkommt.
Was zusätzlich noch sein könnte ist, dass die Pulsdauer zu klein ist, in conf_adv 
#define MINIMUM_STEPPER_PULSE auf 2 setzen.

Gestern hatte einer hier das selbe Problem^^

Der atmega2560 packt max. 40.000 Steps pro Sekunde.
Beispiel 1: 80steps/mm, Geschwindigkeit 100mm/s --> 80*100=8.000steps/s < 40.000--> OK
Beispiel 2: 640steps/mm, Geschwindigkeit 100mm/s --> 640*100=64.000steps/s > 40.000 --> NICHT OK

rt1970, 2018-03-09:
Och schade...
ABER: #define MINIMUM_STEPPER_PULSE stand auf 0
Ich setze die mal auf 2 und teste.
Die Geräusche haben sich tatsächlich nach Überlastung angehört. Hatte sogar 1x ein Reboot.
Warum würde dann aber ein 2130 Stepper am 8bit Board laufen? Andere Ansteuerung/Verarbeitung?

8Roman8, 2018-03-09:
TMC2130 kann mit nativen 256 microsteps angesteuert werden, das machen aber die wenigsten.
Man steuert die treiber mit zb 16 microsteps an und diese werden im Treiber intern auf 256 interpoliert um die Motoren leise zu machen.

rt1970, 2018-03-09:
Ist der Mega256 wohl doch zu schwach. Keine Änderung. Leider.

8Roman8, 2018-03-09:
Aber dein Hauptproblem wurde gelöst oder?
Vielleicht kannst du dann das issue schließen...

rt1970, 2018-03-09:
Hatte ich ein Problem? Wollte eigentlich nur Ruhe im Wohnzimmer ;-)
Ja hast recht! Problem gelöst und vielen Dank!

closed 2018-03-09 by rt1970