MarlinFirmware/Marlin #10023: [1.1.8]XY Skew Compensation Not Working

gizon12, 2018-03-09:
Hello
I have XY Skew Compensation enabled on my delta. I can see the skew factor with the M852 command(0.07 in my case). But no compensation actually occurs when printing, instead whenever the effector changes direction in the X-Y plane, the effector first moves to the opposite direction and then moves back to the correct direction, almost like a zigzag path. This happens only when printing with a non zero skew factor. I tried changing feedrates, acceleration, jerk values, but to no effect. I tried it with and without bilinear bed leveling, but the same thing happens everytime. Has anyone had this issue before or have any idea what maybe causing this?
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1797044/Configuration.zip)

AnHardt, 2018-03-09:
'XY Skew Compensation' is not for DELTAS.
'XY Skew Compensation' is needed when in a cartesian system X and Y are not 90° to each other.
In DELTA systems there are no cartesian XY-axes. So the error comes from some other errors in the geometry, so the fix for cartesian printers might not work.
Fight the reasons - not the symptoms.

gizon12, 2018-03-09:
So could this be an arm length or tower angle issue? Can Marlin's software compensations, like the one for arm length or tower angle be used to correct it?

thinkyhead, 2018-03-10:
@gizon12 — The [`G33` Delta Auto Calibration](http://marlinfw.org/docs/gcode/G033.html) command is recommended to tune your delta parameters. The most current version is in the `bugfix-1.1.x` branch.

boelle, 2019-02-20:
@gizon12 still having issues?

remember to use latest bugfix 2.0

1.1.8 has died long time ago

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead