MarlinFirmware/Marlin #10024: Anet A8 LCD is garbled after a print

HPLJensen, 2018-03-09:
Hi,

After flashing 1.1.8 to my Anet A8 printer, the LCD screen becomes garbled after the end of a print. I can push a button and it will return to not quite normal. A few places is still garbled.

I believe it to be software related as I can't force the garbling(?) before printing by flexing cables or pushing on things. 

Is this a known issue or is it me doing something wrong? Can I change some setting in one of the files to try and remedy the situation?

Thanks!

- HP

[image]

Bob-the-Kuhn, 2018-03-09:
Try the latest bugfix-1.1.x.  It has a fix for the few places garbled after changing the screen.

It may not take care of the main problem.

rszasz, 2018-03-14:
Is that the reprapdiscount lcd/sd card/rotary encoder? adding a few delays seemed to help mine stop corrupting on my anycubic kossel (ramps1.4 clone, with reprapdiscount clone). Rapidly twirling the encoder while going in and out of menus was the most reliable way I found to get corruption to happen and now it's stable.

In the Configuration.h I replaced

#define REPRAP_DISCOUNT_SMART_CONTROLLER

with

#define REPRAP_DISCOUNT_SMART_CONTROLLER
#define ST7920_DELAY_1 DELAY_0_NOP
#define ST7920_DELAY_2 DELAY_1_NOP
#define ST7920_DELAY_3 DELAY_2_NOP

thinkyhead, 2018-03-14:
We might want to add this (or similar) to the Anet A8 configuration if this is a common issue…
```cpp
#define ST7920_DELAY_1 DELAY_0_NOP
#define ST7920_DELAY_2 DELAY_1_NOP
#define ST7920_DELAY_3 DELAY_2_NOP
```

benlye, 2018-03-14:
The standard Anet A8 display is a 2004 LCD.  These problems are usually caused by interference and some users have had success either separating the ribbon cable or using a ferrite choke either on the LCD cable or 12V power line.

For Anet machines with a graphical display these lines are already present in `pins_ANET_10.h`:
```cpp
    #define ST7920_DELAY_1 DELAY_0_NOP
    #define ST7920_DELAY_2 DELAY_1_NOP
    #define ST7920_DELAY_3 DELAY_2_NOP
```

martin072, 2018-08-12:
I can confirm the Bug. I have a Prusa i3 with Merlin 1.1.8 and noticed a similar garbled screen as on the picture. It starts after a while when printing a relative large print job (about 2 hours printing), screen gets garbled after about 30-45 minutes into the print. The job self completes fine, screen does not get better until a reset.

thinkyhead, 2018-08-15:
@martin072 — How do the proposed solutions work for you?

closed 2018-08-15 by thinkyhead

SissyMissy1973, 2018-08-27:
I just flashed to 1.1.19 from a factory firmware on my Anet A8 (had the printer nearly a year without upgrading the firmware) now I'm getting the garbled display problem. Never happened before. It happens a lot when I'm trying to do bed leveling, and will also sometimes happen in the middle of a print. Will not go back to normal until I reset the controller. I have the 2004LCD display. I also don't think my bed leveling is getting saved to the eeprom when I try to store the changes.

misham73, 2018-08-27:
I just update my factory Anet A8 firmware to Marlin 1.1.19 and now I'm getting the garbled display. It happened several times while trying to do bed leveling and several times during prints. I also don't think my bed leveling data is being saved to the eeprom when storing it. Any further help on what to do to fix this?

Atatoth, 2018-08-27:
I my case I touch one of back nuts and bolts on LCD maybe bad gnd

thinkyhead, 2018-08-30:
Try tweaking the LCD timings. The default values set in the Anet pins file are:
```cpp
    #ifndef ST7920_DELAY_1
      #define ST7920_DELAY_1 DELAY_NS(0)
    #endif
    #ifndef ST7920_DELAY_2
      #define ST7920_DELAY_2 DELAY_NS(63)
    #endif
    #ifndef ST7920_DELAY_3
      #define ST7920_DELAY_3 DELAY_NS(125)
    #endif
```
You can override these by adding lines to your `Configuration.h` file such as:
```cpp
#define ST7920_DELAY_1 DELAY_NS(63)
#define ST7920_DELAY_2 DELAY_NS(100)
#define ST7920_DELAY_3 DELAY_NS(125)
```
If you find values that work well, let us know and we'll adjust the defaults for your display model.

betavr, 2019-01-29:
I have the same problem with Tronxy X1 on the 1.1.x branch. Screen gets garbled sometimes, but can be fixed by pressing a button.

AnHardt, 2019-01-29:
What you see is caused by EMV and or electrostatic effects. This is seen quite often in unshielded environments. It's not caused by software. All you can do is reinitalising and reprinting the displays content. That should already be done when you switch from status screen to the menus.
Also a grounded case for the display may help.

@TheOtherSupporters
Here we have the character-displays (HD44780). This is completely unrelated to the graphical ST7920 displays. Altering the delays for that is useless.

HPLJensen, 2019-01-29:
I simply don't believe it's only to do with the shielding. The problem never occurred with the stock firmware. For me, it happened immediately after installing Marlin. Thus it must, at the very least, be the Firmware doing something differently that means the shielding is inadequate for that particular Firmware. Otherwise it should happen with stock firmware as well.

martin072, 2019-01-30:
As much as I would like to agree it could be a shielding/electro static issue, it stil doesn't explain why I've never seen it on my original firmware. Further more, browsing the menu does indeed restore the display, however some characters are still garbled. Please see attached picture. This was shot when the whole display was garbled, then entering a menu option and going back to the main screen. As you can see some symbols are still garbled..

[image]

AnHardt, 2019-01-30:
This is entirely different from the picture above yours. 
Here only some of the self defined symbols are garbed. (Thermometer and °)
That is likely to be a software problem.

Begin with updating to a recent Marlin.

martin072, 2019-01-30:
@AnHardt As mentioned, the screen was fully garbled before (like the other picture). Going into a menu and back will restore the display, but not in full. Hence the discussion if this is a software problem and not shielding issue.

AnHardt, 2019-01-30:
Status screen and menus do define different 'graphics' for the 8 custom characters.
But 'Thermometer' and '°' they have in common. Likely they are not retransfered to the display when switching between the modes.

AnHardt, 2019-01-30:
Looks as if fixed with #9986, ten month ago.

perplex51, 2019-04-17:
so is there a fix?

girs1982, 2019-09-14:
try remove sd card and reboot,in my anet it was help

girs1982, 2019-09-14:
> so is there a fix?

try remove sd card and reboot,in my anet it was help

bera99, 2020-01-29:
[image]

Come posso risolvere?