MarlinFirmware/Marlin #10025: 2.0 BugFix

jacobsokolove, 2018-03-09:
### Description

Compiling Never Completes

### Steps to Reproduce

1. Downloaded the Newest 2.0 Bugfix
2. Configured Everything (namely, enabled the use for an inductive sensor, sw spi, software serial, tmc 2130, 2208
3. Tried to compile and upload

I'm not sure why it won't compile. I'm using a standard ramps board and mega combo.

[Marlin 2.0 Hypercube - Copy.zip](https://github.com/MarlinFirmware/Marlin/files/1797636/Marlin.2.0.Hypercube.-.Copy.zip)

Spawn32, 2018-03-09:
You have to poste the compile errors, if not it's hard to help you for the dev's..

jacobsokolove, 2018-03-09:
There aren't any errors. It never finishes compiling. I waited for about 40 minutes and it never finished compiling. I checked and I never had any issues with previous versions of marlin (even on the same device I'm using to upload everything).

Studiodyne, 2018-03-09:
delete temp folder

thinkyhead, 2018-03-10:
What environment are you using to compile with?

thinkyhead, 2018-03-10:
What environment are you using to compile with? I just compiled your project with Arduino IDE 1.8.5 and had no issues.

jacobsokolove, 2018-03-10:
I was using Arduino IDE 1.8.5 in windows 10 on a microsoft surface. I'll try deleting my temp file to see if that would help, but why would that fix it? what causes that issue?

Bob-the-Kuhn, 2018-03-10:
Most likely it's just another Microsoft bug.  Windows has a reputation for things like that.

jacobsokolove, 2018-03-10:
Honestly, this is the first time I've had a windows issue. interesting. If linux was better supported on the surface, I'd use it.

closed 2018-03-10 by jacobsokolove