MarlinFirmware/Marlin #10026: LCD Garbled - 1.1.x  "bugfix" branch

dcwalmsley, 2018-03-09:
BUG: LCD display is garbled after compiling latest bugfix 1.1.x (9 March 2018).  See attached zipped folder for pic and files.

1. EinsyRambo motherboard with RepRapDiscount FULL GRAPHIC Smart Controller on a custom built CoreXY HyperCube with Prusa MMU 4-extruder kit and Filament Sentinel.

r,
Doug W.

[Marlin-bugfix1.1.x 9Mar18.zip](https://github.com/MarlinFirmware/Marlin/files/1798193/Marlin-bugfix1.1.x.9Mar18.zip)

AnHardt, 2018-03-09:
Too fast. Add delays.

thinkyhead, 2018-03-10:
Try playing with the ST7920 delays. For example, a pretty long set of delays would be…
```cpp
#define ST7920_DELAY_1 DELAY_5_NOP
#define ST7920_DELAY_2 DELAY_5_NOP
#define ST7920_DELAY_3 DELAY_5_NOP
```

dcwalmsley, 2018-03-11:
That did the trick!  Display is readable now.

I uncommented those three lines in the ultracd_st7920_u8glib_rrd.h file and 
changed them to 5's as you mentioned.

Thanks for the help.

R
Doug W

thinkyhead, 2018-03-11:
Try lower numbers than 5, until you find the lowest values that are needed to prevent the garbled display. Recommended because higher numbers actually slow down the display.

closed 2018-03-11 by thinkyhead

thinkyhead, 2018-03-11:
They can also be defined in your `Configuration.h` file, if you prefer.

dcwalmsley, 2018-03-11:
Appreciate the suggestion.