MarlinFirmware/Marlin #10027: Serial communication - checksum missmatch

8Roman8, 2018-03-09:
Since I updated the FW to the newesr bugfix2.0.x branch today, I had about 6 times the problem that the print was aborted or the printer didn't do anything anymore...

I tried to go down with the baudrate down to 57600 but this doesnt help. (octoprint)

Till the update I never had the problem, Maybe I overseen something but all connections looks fine.

> | Send: N6323 G1 X119.575 Y122.250 E0.3404 F2377*10
> | Recv: ok N6320 P6 B1
> | Send: N6324 G1 E-3.0000 F1650*42
> | Recv: Error:checksum mismatch, Last Line: 6322
> | Recv: Resend: 6323
> 

After resend nothing happens.

[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1798186/Configuration.zip)

8Roman8, 2018-03-09:
I will try to reinstall octoprint, maybe this solve this.

8Roman8, 2018-03-10:
I tried to reinstall octoprint but this doesn't solved this issue... I searched for similar issues on octoprint github page and I found something... maybe this can help..

"Your printer requests a resend here, but never "clears" OctoPrint for sending, causing the timeout and the following issues - the ok after the resend request is missing. That's a bug in the printer's firmware."

Link: https://github.com/foosel/OctoPrint/issues/1817

thinkyhead, 2018-03-11:
This has already been patched.

closed 2018-03-11 by thinkyhead