MarlinFirmware/Marlin #10028: [2.0.x] Strange Z home and Broken ABL Probing with Latest bugfix-2.0.x

benlye, 2018-03-09:
Just grabbed the latest bugfix-2.0.x build and am seeing some weird behaviour.  RAMPS setup with Bilinear bed-leveling enabled and an inductive probe.

**G28**
Before, when homing Z the probe would double-touch then move back up a few mm.  Now, there is no final upward movement and the probe stays at the height where it stops on the second downward movement (the point where the probe triggers).

**G29**
My probing grid is 7x7.  As of the latest build the probe stops at the last point in the 6th row, completely skipping the final row.  Afterwards `M420` shows this:
```
Send: M420 V
Recv: echo:M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6
Recv:  0 -0.010 +0.070 +0.107 +0.177 +0.232 +0.295 +0.242
Recv:  1 -0.153 -0.075 -0.013 +0.055 +0.105 +0.125 +0.112
Recv:  2 -0.240 -0.125 -0.063 +0.007 +0.052 +0.095 +0.020
Recv:  3 -0.275 -0.188 -0.110 -0.038 -0.000 +0.002 -0.028
Recv:  4 -0.243 -0.135 -0.073 -0.003 +0.030 +0.055 -0.035
Recv:  5 -0.168 -0.095 -0.035 +0.022 +0.050 +0.042 +0.002
Recv:  6  =====  =====  =====  =====  =====  =====  =====
```

I've captured debug logs of both.
[G28.log](https://github.com/MarlinFirmware/Marlin/files/1798475/G28.log)
[G29.log](https://github.com/MarlinFirmware/Marlin/files/1798476/G29.log)

benlye, 2018-03-09:
Looking at the G29 log I see the final move is this:
```
2018-03-09 21:09:51,558 - Recv: >>> probe_pt(15.00, 183.00, no stow)
2018-03-09 21:09:51,563 - Recv:   current_position=(40.00, 196.00, 3.93) :
2018-03-09 21:09:51,568 - Recv:   current_position=(40.00, 196.00, 3.93) : set_probe_deployed
2018-03-09 21:09:51,570 - Recv: deploy: 0
2018-03-09 21:09:51,574 - Recv: >>> do_blocking_move_to(40.00, 196.00, 3.93)
2018-03-09 21:09:51,577 - Recv: <<< do_blocking_move_to
2018-03-09 21:09:51,581 - Recv:   current_position=(40.00, 196.00, 3.93) : > probing complete
2018-03-09 21:09:51,586 - Recv:   current_position=(40.00, 196.00, 3.93) : bracket_probe_move
```
But it doesn't try to make the move to (15,183); it is stopped at the preceding location, (15,155).

Config files:
[Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1798519/Marlin.zip)

MoellerDi, 2018-03-09:
I'm having a similar issue with bugfix-1.1.x (see #10029). Could it be related to #10020/#10019?

benlye, 2018-03-10:
Fixed by 2908e76151f894bae27f6f455cfbea51e2dc1e41.

closed 2018-03-10 by benlye