MarlinFirmware/Marlin #10029: G29 seems be broken after PR #10019

MoellerDi, 2018-03-09:
G29 seems be broken now after PR #10019 got merged. It seems the probing just stop incomplete in an area where the nozzle normally would needs to leave the bed in order to enable to probe to reach a probing position.

The bed size of my CR-10 is X 300, Y 300 but the nozzle can be moved to X 320, Y 300. So IMHO it should be ok and it was working fine before PR #10019 got merged.

My configurations can be found at https://github.com/MoellerDi/Marlin/tree/bugfix-1.1.x-HD-MKSGen14

As kind of workaround I set the following for now to avoid a situation where the nozzle have to leave the bed and to stop probing incomplete (probe offset is X:-46 Y:-11):
```
  #define LEFT_PROBE_BED_POSITION 50 // (used to be set to 30)
  #define RIGHT_PROBE_BED_POSITION (X_BED_SIZE - LEFT_PROBE_BED_POSITION)
  #define FRONT_PROBE_BED_POSITION 50 // (used to be set to 30)
  #define BACK_PROBE_BED_POSITION (Y_BED_SIZE - FRONT_PROBE_BED_POSITION)
```

Below please find the debug log; please let me know if any more information would be needed.

Recv: echo:M111 S247
Recv: echo:DEBUG:ECHO,INFO,ERRORS,COMMUNICATION,LEVELING
Recv: ok P15 B3
[...]
Send: G28
Recv: echo:G28
Recv: >>> gcode_G28
Recv: Machine Type: Cartesian
Recv: Probe: FIX_MOUNTED_PROBE
Recv: Probe Offset X:-46 Y:-11 Z:-1.15 (Left-Front & Below Nozzle)
Recv: Auto Bed Leveling: BILINEAR (disabled)
Recv: 
Recv:   current_position=(196.00, 161.00, 1.15) : setup_for_endstop_or_probe_move
Recv: > endstops.enable(true)
Recv: Raise Z (before homing) to 4.00
Recv: >>> do_blocking_move_to(196.00, 161.00, 4.00)
Recv: <<< do_blocking_move_to
Recv: >>> homeaxis(X)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(X, -480.00, 0.00)
Recv:   current_position=(0.00, 161.00, 4.00) : sync_plan_position
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: <<< do_homing_move(X)
Recv: >>> set_axis_is_at_home(X)
Recv: For X axis:
Recv:  home_offset = 0.00
Recv:  position_shift = 0.00
Recv:  soft_endstop_min = 0.00
Recv:  soft_endstop_max = 320.00
Recv: > home_offset[X] = 0.00
Recv:   current_position=(0.00, 161.00, 4.00) :
Recv: <<< set_axis_is_at_home(X)
Recv:   current_position=(0.00, 161.00, 4.00) : sync_plan_position
Recv:   current_position=(0.00, 161.00, 4.00) : > AFTER set_axis_is_at_home
Recv: <<< homeaxis(X)
Recv: >>> homeaxis(Y)
Recv: Home 1 Fast:
Recv: >>> do_homing_move(Y, -450.00, 0.00)
Recv:   current_position=(0.00, 0.00, 4.00) : sync_plan_position
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
[...]
Recv: echo:busy: processing
Recv: <<< do_homing_move(Y)
Recv: >>> set_axis_is_at_home(Y)
Recv: For Y axis:
Recv:  home_offset = 0.00
[... 213 lines cut ...]
Recv: >>> do_blocking_move_to(286.00, 41.00, 2.38)
Recv: <<< do_blocking_move_to
Recv:   current_position=(286.00, 41.00, 2.38) : > probing complete
Recv:   current_position=(286.00, 41.00, 2.38) : clean_up_after_endstop_or_probe_move
Recv: <<< G29
Recv: X:286.00 Y:41.00 Z:2.38 E:0.00 Count X:22880 Y:3280 Z:952
Recv: ok P15 B3
[...]
Send: M420 V
Recv: echo:M420 V
Recv: Bilinear Leveling Grid:
Recv:       0      1      2      3      4      5      6      7      8
Recv:  0 -0.200 -0.135 -0.115 -0.023 +0.013 +0.107 +0.148 +0.230  =====
Recv:  1  =====  =====  =====  =====  =====  =====  =====  =====  =====
Recv:  2  =====  =====  =====  =====  =====  =====  =====  =====  =====
Recv:  3  =====  =====  =====  =====  =====  =====  =====  =====  =====
Recv:  4  =====  =====  =====  =====  =====  =====  =====  =====  =====
Recv:  5  =====  =====  =====  =====  =====  =====  =====  =====  =====
Recv:  6  =====  =====  =====  =====  =====  =====  =====  =====  =====
Recv:  7  =====  =====  =====  =====  =====  =====  =====  =====  =====
Recv:  8  =====  =====  =====  =====  =====  =====  =====  =====  =====
Recv: 
Recv: echo:Bed Leveling Off
Recv: echo:Fade Height 10.00
Recv: ok P15 B3

VanessaE, 2018-03-09:
Broken for me as well.  My first probe point has the nozzle and probe fully over the bed, but it just stops there after probing that spot instead of moving to the next point, which has the nozzle just off the bed (but the probe squarely over its usual spot).

VanessaE, 2018-03-09:
`git revert fa12866de32c001e08cdda340d6d7006e8001c82` is sufficient to allow `G29` to complete.  Can't say what other effect that will have, though.

thinkyhead, 2018-03-10:
Sorry, got my variables backwards. The `position_is_reachable_by_probe` function is now corrected!

commit 6d9ea7e1 (2018-03-10): Fix position_is_reachable_by_probe typo, add comment

Fix #10029
commit 0d13c5d0 (2018-03-10): Fix position_is_reachable_by_probe, add comments

Fix #10029
commit 2908e761 (2018-03-10): Fix position_is_reachable_by_probe typo, add comment

Fix #10029
closed 2018-03-10 by thinkyhead