MarlinFirmware/Marlin #10033: [bugfix 2.0x ] Arduino DUE+RAMPS-FD V2+ LCD 2004 ,Power on initialization takes 8 seconds

cjsoong, 2018-03-10:
2018-03-10 download bugfix 2.0x ,
Power on takes 8 seconds to start displaying the boot screen.
2018-3-05 download is  ok
Is it caused by the EEROM i2c  or LCD SPI correction file?

Bob-the-Kuhn, 2018-03-10:
I'm nor seeing that when I compile with Arduino 1.9.0 beta.

I can't test with PlatformIO.  They did an update last night and now the LiquidCrystal library is gone.

We did add the following last night to V2:
```
#ifndef PS_ON_PIN
  #define PS_ON_PIN        12
#endif

#ifndef FILWIDTH_PIN
  #define FILWIDTH_PIN      5   // Analog Input on AUX2
#endif
```

Bob-the-Kuhn, 2018-03-10:
There is definitely something strange going on with the native USB port.  Try disconnecting it & see if things return to normal.

Last night's changes made the USB mass storage work on my Windows machine.  It uses the native USB port.

Today sometimes I can't flash the DUE and sometimes it won't come up.  I need to investigate further to see what causes this.

Bob-the-Kuhn, 2018-03-10:
I've just got a minimal configuration.  V2, LCD 7 SD card enabled.

Bob-the-Kuhn, 2018-03-10:
I've just reverted PR #9961.  Sometimes i have to re-program the bootloader.

ejtagle, 2018-03-11:
Just an small update: When using the USB stack supplied by Atmel, the native port can´t be used to reprogram the firmware anymore. But, the other port (the serial console one) still works as intended. 
Using the native port to reflash firmware is, in fact, a hack done by the arduino guys... That is why the Atmel supplied USB stack (required to support USB mass storage and SerialOverUSB connections with proper USB handshake, does not support flash reprogramming...

ejtagle, 2018-03-11:
I had never had any issues with the bootloader or anything. Always used the Programming Port. My board does not have any kind of eeprom (neither I2c or SPI).
If i had to guess, i´d say something in the initialization or read of the EEPROM is taking too long and the watchdog is kicking in and resetting the part. Eventually, it manages to read the eeprom and execution continues without issues.

Bob-the-Kuhn, 2018-03-11:
As best I can tell the new USB code doesn't allow the native port to program the DUE or to connect to the host.

As best I can tell the DUE FLASH gets corrupted if you
1 Download an image with the SD card enabled.
2 Connect the native port (the USB drive then pops up).
3. Download another image.  The download fails because the com port isn't there.  The only way to recover is to use an external programmer. 

I'm trying to isolate the offending changes but compiles take about 5 minutes on my PC so it's slow going.

Bob-the-Kuhn, 2018-03-11:
It looks like the PR #9961 changes are NOT the culprit.  I just played with the reverted bugfix-2.0.x and it has the same issue.

On Windows if the native port is plugged into the computer when doing a download the download will fail and the controller will be dead.

ejtagle, 2018-03-11:
#10063 should reenable the native USB port to reprogram the part (it works for me)

ejtagle, 2018-03-14:
I've been experiencing lately the same boot delays as @cjsoong  ... In my case, seems to be related to the SD card initialization... My initial guess is that the Initialization loop has a 1 second timeout. And the Watchdog kicks in and resets the target ... I will probably investigate it .... 👍

p3p, 2018-03-14:
@ejtagle To fix a recent watchdog timeout issue with slow UBL startup (outputs the active mesh over serial) the watchdog is not enabled until the end of `setup()`. (if you do have UBL enabled though you will have a significantly longer boot delay)

ejtagle, 2018-03-15:
I think this could be related to @Bob-the-Kuhn findings (issue #10106 ) 👍

ejtagle, 2018-03-24:
i think this issue should be solved. Please retest with latwst version.!+

eyal0, 2018-04-15:
I think that I'm seeing this issue.  When OctoPrint tries to connect to the printer, the printer stalls a very long time.  Eventually OctoPrint gives up with a timeout message and then the printer works fine.

thinkyhead, 2018-04-20:
Is there any log output before the stall? What about other hosts?

eyal0, 2018-04-27:
@thinkyhead I figured it out. OctoPrint starts with a M20 and that was going into an endless loop, I think. There were hundreds of lines about "can't open subdir S3D". I reformatted the SDcard and the problem went away.

The only thing that might be worth looking at is that I had an s3d directory on both the SDcard and the Marlin storage. I don't recall if they had the same capitalization, I think that they didn't. Maybe that confused Marlin when traversing the directory structure. If I get around to.it, I'll see if I can reproduce the bug.

closed 2018-04-29 by thinkyhead