MarlinFirmware/Marlin #10034: [Linear Advance 1.5]  : Suggestion ( Not an issue )

Studiodyne, 2018-03-10:
@Sebastianv650 @thinkyhead 

Hi ... I would suggest you something easy ....
I hate the value of Linear Advance ,  to set a value under 1 or 2 and have tons of decimal , not really ... a pleasure
Is it possible to use an integer value ' by multiplying by 100 ' to have   a beautifful setting that go from zero to 200 for example ...
To set this 0.06 doesn't really cool , and doesn't give good feeling with this feature

Just suggestion

Sebastianv650, 2018-03-10:
I see the new unit as a big step forward because it not longer "a magic factor somewhere between 100 and 200". It has a real unit, which is: mm of filament compression needed for an extrusion speed of 1mm/s (at the given extruder setup) or short [mm/mm/s].
The funny thing here is that this results in the unit second for K, K=0.22s :)

So using this unit, it's easy to compare K for different printers or filaments, as this compression length value is something you can imagine and you have a feeling for, just like km/h or meters.
Of course we could also use µm/mm/s which would mean a K of 0.22 would become 220, but mm is much more common and usualy values in this magnitude are not stated in µm.

Studiodyne, 2018-03-10:
100 % ok with you
But just a question of ' feeling ' , nothing very important

Thks

closed 2018-03-10 by Studiodyne