MarlinFirmware/Marlin #10041: [bugfix-1.1.x] Anet LCD 2004 Up and Down buttons are reversed

MikeLud, 2018-03-10:
The current bugfix-1.1.x that  I download today (72df10c) has the LCD 2004 Up and Down buttons reversed.

Thanks,
Mike

Bob-the-Kuhn, 2018-03-10:
Please post your configuration.h file so we know for sure what LCD and motherboard you are using.

MikeLud, 2018-03-10:
Anet A8 configuration.h file 
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1799689/Configuration.zip)

Bob-the-Kuhn, 2018-03-10:
I can't find where those buttons were changed after the LCD was added to Marlin.

Go into your **pins_ANET_10.h** file and make the changes.  Once you've tested the changes please post the before & after files.

I'm a little concerned that there is something unique about your setup and what works for you may not work for others.  I'll propose the changes but wait to see if anyone objects before merging.

benlye, 2018-03-10:
The min/max ADC values for each button on the 5-button panel are defined in ultralcd.cpp:
https://github.com/MarlinFirmware/Marlin/blob/bugfix-1.1.x/Marlin/ultralcd.cpp#L5559

@MikeLud You can enable `ADC_KEYPAD_DEBUG` by adding `#define ADC_KEYPAD_DEBUG` somewhere in configuration.h to see the actual value reported when a button is pushed.

I'll dust off my A8 board and LCD tomorrow and see if I can repro the issue.

MikeLud, 2018-03-11:
Before the below commit the buttons works fine, looks like this is what causes the buttons to reverse https://github.com/MarlinFirmware/Marlin/commit/70e9c07b2d888aba3e24aefac174e246e620d2d5

thinkyhead, 2018-03-11:
I've just patched `scroll_screen` to fix the menu direction and multiplier, so this should now be fixed.

benlye, 2018-03-11:
That fixed it for me.

closed 2018-03-11 by thinkyhead

MikeLud, 2018-03-11:
Just loaded and buttons are back to normal. 

Thanks

commit 51ebdf3b (2018-03-11): Finish scroll_screen with menu direction/multiplier

Fixes #10041
commit eb6f4437 (2018-03-11): Finish scroll_screen, fixing direction/multiplier

Fixes #10041