MarlinFirmware/Marlin #10042: [2.0] undefined reference to `GcodeSuite::M666()

Studiodyne, 2018-03-10:
Look in my Pr ' Tool migration '
Compilation ok at home
But travis  give red light because of undefined reference to `GcodeSuite::M666()

Look TRAVIS CHECK in my pr , you will have all informations

close it after read

8Roman8, 2018-03-10:
I get an error while compilation.

```
In function `GcodeSuite::process_next_command()':

cc5kNRYm.ltrans0.o:(.text+0xa68): undefined reference to `GcodeSuite::M666()'

collect2.exe: error: ld returned 1 exit status

exit status 1
```

Studiodyne, 2018-03-10:
Don't worry , thinkyhead , the author of the last commits , will repair
Just inform him , the max as possible
like me 
and wait 
I would love to repair myself  but  .... Too deep  , only the author will solve this quickly

8Roman8, 2018-03-11:
My printer is cartesian as well ^^

thinkyhead, 2018-03-11:
> I would love to repair myself but

If you see a compile error lingering for more than 24 hours then it's worth posting an issue.

Also, if Travis CI is already barking about it ❌ , then it's obvious and posting about the error is redundant.

closed 2018-03-11 by thinkyhead