MarlinFirmware/Marlin #10043: TMC2130 Software-SPI: driver-failure during print

LichtiMC, 2018-03-10:
For testing purposes I have only fitted the X-axis with an TMC2130-driver.

I'm using Arduino DUE SAM3X8E with SMART_RAMPS board, SPI is configured like this:
```  #define TMC_USE_SW_SPI
  #define TMC_SW_MOSI       51
  #define TMC_SW_MISO       50
  #define TMC_SW_SCK        52
```
With this configuration printing starts fine, but after a random time the tmc-driver fails and just stands still. I think I can rule out temperature problems as my H-Bot printer is using 400ma stepper motors. The tmc doesn't even get warm, the motors get warm to approximately 40 degrees:
```
  #define X_CURRENT          400  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256
```

This happened with the 2 last prints:
[image]

Could this problem arise due to a bug in the software?
How could I debug the failure?
Thanks

8Roman8, 2018-03-10:
A had a similar issue with tmc2208, my acceleration was to high and the motor driver shuts down...

LichtiMC, 2018-03-10:
These are my acceleration-values:
```
/**
 * Default Max Acceleration (change/s) change = mm/s
 * (Maximum start speed for accelerated moves)
 * Override with M201
 *                                      X, Y, Z, E0 [, E1[, E2[, E3[, E4]]]]
 */
#define DEFAULT_MAX_ACCELERATION      { 1000, 1000, 100, 10000 }

/**
 * Default Acceleration (change/s) change = mm/s
 * Override with M204
 *
 *   M204 P    Acceleration
 *   M204 R    Retract Acceleration
 *   M204 T    Travel Acceleration
 */
#define DEFAULT_ACCELERATION          1000    // X, Y, Z and E acceleration for printing moves
#define DEFAULT_RETRACT_ACCELERATION  1000    // E acceleration for retracts
#define DEFAULT_TRAVEL_ACCELERATION   1000    // X, Y, Z acceleration for travel (non printing) moves
```
What values are you using?
Thanks.

8Roman8, 2018-03-10:
Hm... I think 1000 isn't so high... 
Do you use "high" voltage motors? 400ma looks to me too little for a standard motor, maybe yor psu voltage to little for the combination of your motor and the driver

LichtiMC, 2018-03-10:
Don't think so.
I use 24V, A4988 Stepper Motor Driver and these motors for months now without a problem.
(It's the Creality Ender 4 printer.)
The only problem is with the TMC2130 driver.

teemuatlut, 2018-03-10:
Try to enable monitoring so that Marlin will tell you when the driver reports an error condition.
You can decide for yourself whether or not you want to automatically stop the print.

LichtiMC, 2018-03-10:
I have (and always had) following options set:
```C++
#define MONITOR_DRIVER_STATUS

  #if ENABLED(MONITOR_DRIVER_STATUS)
    #define CURRENT_STEP_DOWN     50  // [mA]
    #define REPORT_CURRENT_CHANGE
    #define STOP_ON_ERROR
  #endif
```
Shouldn't it already have stopped the print with these settings, instead of going ahead printing without the X-axis?

LichtiMC, 2018-03-11:
@#%$§!
[image]
```
Send: M122
Recv: 		X
Recv: Enabled		false
Recv: Set current	400
Recv: RMS current	718
Recv: MAX current	1012
Recv: Run current	12/31
Recv: Hold current	6/31
Recv: CS actual		0/31
Recv: PWM scale	0
Recv: vsense		0=.325
Recv: stealthChop	false
Recv: msteps		256
Recv: tstep		17213
Recv: pwm
Recv: threshold		0
Recv: [mm/s]		-
Recv: OT prewarn	false
Recv: OT prewarn has
Recv: been triggered	false
Recv: off time		0
Recv: blank time	16
Recv: hysterisis
Recv: -end		-3
Recv: -start		1
Recv: Stallguard thrs	0
Recv: DRVSTATUS	X
Recv: stallguard
Recv: sg_result		0
Recv: fsactive
Recv: stst
Recv: olb		X
Recv: ola		X
Recv: s2gb
Recv: s2ga
Recv: otpw
Recv: ot
Recv: Driver registers:	X = 0x60:00:00:00
```

8Roman8, 2018-03-11:
Haveyou tired the standalone mode? Maybe this depends on the driver itselv?

thinkyhead, 2018-03-11:
You described the drivers as staying pretty cool. But it can never hurt to put a fan directly over them. They may jump in temperature under certain conditions. Also, it's not a bad idea to have some spares and swap them out to see if the problem is reproducible with 2 or more drivers.

LichtiMC, 2018-03-11:
Thanks, now I'm testing HW-SPI and started printing 30 min. ago.
I already have a fan for the board.

And as I have "#define MONITOR_DRIVER_STATUS" enabled it should give me some error message over serial or at least stop printing, shouldn't it?

towe96, 2018-12-13:
Dear LichtiMC, have you had success with your new drivers?
I've just bought 5 TMC2130's from Watterot which experience the same issue you described - only that for me, X Y and E stop together all at the same time. I too have the "E0:00:00:00" driver register contents, which do sort of indicate an odd reset or communication issue.

I've tried reducing acceleration and jerk, different currents (which didn't work) and even disabling Stealthchop - which did work, but I could've just kept using the A4988s at that point.
I'm running them on 24V as well with a step-up converter and 100 µF capacitors between VM and GND, directly on the driver.

Perhaps I'll try the lower SPI speeds intended for SD card functionality and see if that makes a difference. It's odd that the Z and Z2 drivers respond perfectly well though - maybe it's a thermal issue still?
When running only two Chinese Fysetc TMC2130s on X and Y, they didn't have temperature issues at all though, and the heatsink is cold as well. Also, if it were thermally related, I suspect it should've just reduced the current like the Fysetc ones once did.

closed 2019-07-17 by shitcreek