MarlinFirmware/Marlin #10048: Cannot use EEPROM Saving or Loading with Re-arm+Ramps1.4

chrisjenda, 2018-03-11:
### Description

Cannot use EEPROM on Re-Arm+Ramps 1.4 with Commit/Pull: 82a02a3/10044

### Steps to Reproduce

1. [First Step]
Setup Marlin Config as Follows:
//  Set Baud Rate to 230400
//  Set MB to Re-Arm + Ramps EFB
//  Set Filiment Diameter to 1.75
//  Enable ATX Power Supply Control
//  Enable EEPROM
//  Enable REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER
//  Enable Dual Z Stepper drivers (Advanced Config)
2. [Second Step]
Compile Marlin and Upload to Re-Arm
3. [Third Step]
Issue an M502 to reset to hardcoded and then run an M500 to get the error.

**Expected behavior:** [What you expect to happen]
M500 Saves EEPROM Settings and allows an M503 to Load them.

**Actual behavior:** [What actually happens]
When sending an M502+M500 Gcode Command the following is returned:

18:57:34.904 : M502
18:57:35.889 : echo:Hardcoded Default Settings Loaded
18:57:39.627 : M500
18:57:40.270 : write_data(100,268468032,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268468014,1...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268444280,16...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268444300,16...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268444352,16...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268442908,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268444328,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268444324,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268444316,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268444348,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268444320,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268444384,16...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268442836,12...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268468040,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268468028,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268468015,1...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268468016,1...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268468028,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268468028,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268468028,4...)
18:57:40.270 : f_lseek()=9
18:57:40.270 : write_data(102,268468028,4...)
[... 202 lines cut ...]
18:57:40.272 : f_lseek()=9
18:57:40.272 : write_data(102,268468028,4...)
18:57:40.272 : f_lseek()=9
18:57:40.272 : write_data(102,268468028,4...)
18:57:40.272 : f_lseek()=9
18:57:40.272 : write_data(102,268468028,4...)
18:57:40.272 : f_lseek()=9
18:57:40.272 : write_data(102,268468028,4...)
18:57:40.272 : f_lseek()=9
18:57:40.272 : write_data(102,268468028,4...)
18:57:40.272 : f_lseek()=9
18:57:40.272 : write_data(102,268468028,4...)
18:57:40.273 : f_lseek()=9
18:57:40.273 : write_data(102,268468028,4...)
18:57:40.273 : f_lseek()=9
18:57:40.273 : write_data(102,268468028,4...)
18:57:40.273 : f_lseek()=9
18:57:40.273 : write_data(102,268468028,4...)
18:57:40.273 : f_lseek()=9
18:57:40.273 : write_data(100,129384,4...)
18:57:40.273 : f_lseek()=9
18:57:40.273 : write_data(100,268468026,2...)
18:57:40.273 : f_lseek()=9
18:57:40.273 : echo:Settings Stored (2 bytes; crc 0)
18:57:40.273 : Error:EEPROM datasize error.

As3uS, 2018-03-12:
Hi there,
Excactly the same problem here with MKS Sbase + Reprap discount FullGraphic. 
Send M500 with ponterface cause same listing error  write_data(102,268468028,4...)
and Mks Reboot few second after.

Roxy-3D, 2018-03-12:
Do you have an SD Memory card plugged into the Re-ARM board?   That is where the EEPROM data is stored.

apballard, 2018-03-12:
Hi all, I’m having issues myself, but not sure if it’s the same problem.

I can save my settings to sd card, I can initialize it, but in restart my printer will not boot up.  It just sits with blank screen. If I remove SD card or delete eeprom file it works. Not sure if related.

chrisjenda, 2018-03-12:
@Roxy-3D Yes, I have an SD card inserted as that is what I used to flash the marlin bin file.

p3p, 2018-03-12:
@apballard if you have UBL enabled can you try the current bugfix-2.0.x, there was an issue with the watchdog resetting while printing the current UBL mesh over serial during boot, (there has to be a long delay per output line for some reason). 

@chrisjenda I'm not sure whats going on here, well I know it's not managing to open the eeprom.dat file for writing, it stands to reason if the bootloader can flash from the sdcard Marlin should be able to read/write it, but to grasp at straws do you have any other sd cards to test with?

apballard, 2018-03-13:
@p3p I think that might be it, let me try.

chrisjenda, 2018-03-13:
I think it has something to do with an update as I belive pull 8917 works fine, but that is pretty far back.  If I run the Command with debugging enabled it gives "19:11:54.958 : Error:Field esteppers mismatch."  So must be a problem with the esteppers field, whatever that means.

As3uS, 2018-03-13:
May be a datasize error like on the /9240 commit ? 
Saw it have some similar issue on the mks and re-arm ?

Bob-the-Kuhn, 2018-03-14:
I'm not able to reproduce your problem.  I just grabbed the latest bugfix and made the changes as above,  I then did the M502, M500 sequence with no complaints.  I then did a couple of M200, M500, M501 cycles without any issues.  Power cycling didn't affect the EEPROM contents (same CRC)..  

Please post your config files & I'll try again.

This problem may have been fixed.  A couple of days ago I couldn't drop a new BIN file onto the USB mass storage window and get it to load into FLASH.  Today all is well.

This is on a Win10 machine.

As3uS, 2018-03-14:
In my case, when the printer was connected on usb, windows10 can't read the SD, 
A window opened but said unable open, unable to mount. 
I downloaded my current bugfix on 03/11/18
Maybe caused by Fullgraphic LCD ?

chrisjenda, 2018-03-14:
I have tried with the latest commit and the problem still persists,  I have tried multiple SD cards and also with different types.  I have provided the config files for which I have used during these tests.
https://ufile.io/pwh1n

This is the log created from issuing the following commands (M502, M500, M501) in that order.

11:27:28.540 : echo:Hardcoded Default Settings Loaded
11:27:32.940 : write_data(100,268468032,4...)
11:27:32.940 : f_lseek()=9
11:27:32.940 : Error:Field esteppers mismatch.
11:27:32.940 : write_data(102,268468014,1...)
11:27:32.940 : f_lseek()=9
11:27:32.940 : write_data(102,268444292,16...)
11:27:32.940 : f_lseek()=9
11:27:32.941 : write_data(102,268444312,16...)
11:27:32.941 : f_lseek()=9
11:27:32.941 : write_data(102,268444364,16...)
11:27:32.941 : f_lseek()=9
11:27:32.941 : write_data(102,268442920,4...)
11:27:32.941 : f_lseek()=9
11:27:32.941 : write_data(102,268444340,4...)
11:27:32.941 : f_lseek()=9
11:27:32.941 : write_data(102,268444336,4...)
11:27:32.941 : f_lseek()=9
11:27:32.941 : write_data(102,268444328,4...)
11:27:32.941 : f_lseek()=9
11:27:32.941 : write_data(102,268444360,4...)
11:27:32.941 : f_lseek()=9
11:27:32.941 : write_data(102,268444332,4...)
11:27:32.941 : f_lseek()=9
11:27:32.942 : write_data(102,268444396,16...)
11:27:32.942 : f_lseek()=9
11:27:32.942 : write_data(102,268442848,12...)
11:27:32.942 : f_lseek()=9
11:27:32.942 : write_data(102,268468040,4...)
11:27:32.942 : f_lseek()=9
11:27:32.942 : write_data(102,268468028,4...)
11:27:32.942 : f_lseek()=9
11:27:32.942 : write_data(102,268468015,1...)
11:27:32.942 : f_lseek()=9
11:27:32.942 : write_data(102,268468016,1...)
11:27:32.942 : f_lseek()=9
11:27:32.942 : write_data(102,268468028,4...)
11:27:32.942 : f_lseek()=9
11:27:32.942 : write_data(102,268468028,4...)
11:27:32.942 : f_lseek()=9
11:27:32.943 : write_data(102,268468028,4...)
11:27:32.943 : f_lseek()=9
11:27:32.943 : write_data(102,268468028,4...)
11:27:32.943 : f_lseek()=9
11:27:32.943 : write_data(102,268468028,4...)
11:27:32.943 : f_lseek()=9
11:27:32.943 : write_data(102,268468028,4...)
11:27:32.943 : f_lseek()=9
11:27:32.943 : write_data(102,268468028,4...)
11:27:32.943 : f_lseek()=9
11:27:32.943 : write_data(102,268468028,4...)
11:27:32.943 : f_lseek()=9
11:27:32.943 : write_data(102,268468028,4...)
11:27:32.943 : f_lseek()=9
11:27:32.943 : write_data(102,268468044,4...)
11:27:32.943 : f_lseek()=9
11:27:32.944 : write_data(102,268468028,4...)
11:27:32.944 : f_lseek()=9
11:27:32.944 : write_data(102,268468028,4...)
11:27:32.944 : f_lseek()=9
11:27:32.944 : write_data(102,268468028,4...)
11:27:32.944 : f_lseek()=9
11:27:32.944 : write_data(102,268468028,4...)
11:27:32.944 : f_lseek()=9
11:27:32.944 : write_data(102,268468028,4...)
[... 180 lines cut ...]
11:27:32.955 : f_lseek()=9
11:27:32.955 : write_data(102,268468028,4...)
11:27:32.955 : f_lseek()=9
11:27:32.955 : write_data(102,268468028,4...)
11:27:32.955 : f_lseek()=9
11:27:32.955 : write_data(102,268468028,4...)
11:27:32.955 : f_lseek()=9
11:27:32.955 : write_data(102,268468028,4...)
11:27:32.955 : f_lseek()=9
11:27:32.955 : write_data(102,268468028,4...)
11:27:32.955 : f_lseek()=9
11:27:32.956 : write_data(102,268468028,4...)
11:27:32.956 : f_lseek()=9
11:27:32.956 : write_data(102,268468028,4...)
11:27:32.956 : f_lseek()=9
11:27:32.956 : write_data(102,268468028,4...)
11:27:32.956 : f_lseek()=9
11:27:32.956 : write_data(102,268468028,4...)
11:27:32.956 : f_lseek()=9
11:27:36.186 : read_data(100,268468028,4...)
11:27:36.187 : f_lseek()=9
11:27:36.187 : read_data(100,268468018,2...)
11:27:36.187 : f_lseek()=9
11:27:36.187 : echo:EEPROM version mismatch (EEPROM=? Marlin=V52)
11:27:36.187 : echo:Hardcoded Default Settings Loaded

p3p, 2018-03-14:
I'm afraid I can't reproduce this with any of my sdcards by using fresh clone of bugfix-2.0.x with your config files applied, I presume you can access the USB Mass Storage device to flash new firmware?, (so the cards are working) is the eeprom.dat file ever created?

As3uS, 2018-03-14:
It's possible that issue was caused by FullGraphic LCD ?
This LCD screen can't provide second SD card access, EXP2 pin use same canal from microSD, conflict ? 
@p3p what lcd have you ?

p3p, 2018-03-14:
I use a `REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER`, The onboard sdcard is used for firmware flashing and the eeprom data and is completely isolated from other spi communications, it cannot currently be used by Marlin for gcode files, the sdcard on the controller can be used as it normally would be.

As3uS, 2018-03-14:
And no i can't access to mass storage
[image]
Maybe smoothiedriver ? But no issue in peripherals

As3uS, 2018-03-14:
i juste dowload firm+u8glib
only set lcd and board in confuguration.h and same issue what commit/issue i should read ?

chrisjenda, 2018-03-15:
@p3p the dat file is not ever getting created with the latest commit,  I have checked commit 1746a7352efb51d44d47ba5e1deb669fa2b2269f and the EEPROM functions normally.  I suspect something from eeprom v48 to 52 has caused the problem.

p3p, 2018-03-15:
@chrisjenda I wish I could reproduce this so I had something to debug, but I cloned bugfix-2.0.x, applied your configs, built and flashed, everything worked as expected. I'l check again but I atm I don't see a problem in the software.

@As3uS I don't have a version, I always test with a clean copy of bugfix-2.0.x, just applying any supplied configs, (being why they are a required part of the issue template)

Bob-the-Kuhn, 2018-03-15:
I get those same errors if I remove the SD card from the RE_ARM board.  It doesn't matter if the SD card is in the LCD controller or not.

If I remember correctly there is a 32G limit on the volume size for the SD card.  I had to format my 64G card as a single 32G volume before it worked.

chrisjenda, 2018-03-15:
I'm at a loss here, the only thing that I can think of is that I had to flash a replacement bootloader to the Re-Arm shortly after I received it as it shorted out causing the bootloader to become corrupted.  I could not find the original bootloader nor could I get it from pannutech so I found that it must just be the same bootloader for the LPC17XX so I downloaded the latest dev version of the DFU bootloader and flashed it.  one of you with the re-arm would want to download your bootloader with flash magic and upload it for me, would you?

Bob-the-Kuhn, 2018-03-15:
Here's a link to the [image of the FLASH](https://www.dropbox.com/s/7l7ku32lfrb5igv/LPC1768_image.zip?dl=0) in my Re_ARM.

It's a binary file for 0 - 0x0007 FFFF.

I don't think the bootloader is the problem.  My understanding is that all of the LP1768 FLASH is written when you upload an image via the SD card on the RE-ARM.  Well, not all, just the part that's actually used.

There isn't a bootloader in FLASH like there is on the AVR chips.  There is a bootloader ROM within the LPC1768 that actually does the writing to FLASH.  If there is valid user code in FLASH on power up or reset then control is passed to the user code and it's the user code that interacts with the boot ROM to download a new image.  If the user code is invalid then the boot ROM is in control and it looks for commands via serial port 0.

As3uS, 2018-03-15:
@p3p , i understand thank, i will give you my MKS configuration tonight.

p3p, 2018-03-15:
@Bob-the-Kuhn There is the smoothieware bootloader in the flash, it uses the first 16KB. When turned on it looks for firmware.bin and flashes it, offset correctly.

@chrisjenda As long as it actually manages to flash the new firmware I don't see the bootloader being the issue, does the firmware.bin file get renamed to firmware.cur?, and can you confirm build date shown at boot (you will probably need to use the hardware serial port to see it, USB takes to long to initialise and connect)

Bob-the-Kuhn, 2018-03-15:
Chris - does the Smoothieware bootloader get overwritten when firmware.bin is loaded?

p3p, 2018-03-15:
No, it puts firmware.bin into flash offset by 16KB, in the linker script we set the flash origin for Marlin to 16KB

As3uS, 2018-03-15:
@p3p  I joint you my configuration files  use frome the last marlin 2.0.x 
Actually SD card Mount but width need format windows popup ? Firmware.bin is correctly rename on first boot.
[Config-Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1816508/Config-Marlin.zip)

Bob-the-Kuhn, 2018-03-15:
Are you putting your SD card into your PC to download the firmware.bin file to it?

On my Win10 PC when I power up my Re_ARM, the first popup says that there's a problem with the drive but then up pops the explorer window for the drive.  I can then copy my firmware.bin file to the SD card via the explorer window.

If Windows isn't popping up the explorer window then there's probably a driver problem.  Win10 is supposed to automatically install the correct driver but I remember problems in that area when I first started using RE_ARM.  Take a look at the [Smoothieware Windows driver page](http://smoothieware.org/windows-drivers) for help on this.

Bob-the-Kuhn, 2018-03-15:
@As3uS - I just compiled your files & uploaded.  Windows popped up the explorer window as usual.

As3uS, 2018-03-15:
[image]
@Bob-the-Kuhn thank for your test.
The smoothiedriver is okay, sdcard mount correctly directly on the computer.

I don't understand ... Will try another computer.
And flash the Smoothieware firm saw if sd mount.

As3uS, 2018-03-15:
Just test, with the smoothieware firmware the SD mount correctly by usb storage mass

Bob-the-Kuhn, 2018-03-15:
@chrisjenda - you'll need to go through the serial 0 port to copy my image into your LPC1768.  BIN files don't contain absolute addresses so putting my image on the SD card & rebooting will just crash your controller.

If I remember correctly something like an FTDI cable is needed to make use of the serial 0 port.

Bob-the-Kuhn, 2018-03-15:
@As3uS - does that mean everything is working for you?

As3uS, 2018-03-15:
@Bob-the-Kuhn nope sir.
->My computer / marlin -> No mounting SD
-> My computer / Smoothie firmware -> sd mount correcty
-> Other W10 computer (i stole my wife's laptop ^^) / Marlin ->sd mount correctly

So, it's my computer causing issue on USB storage Mass. 
Probably driver if i follow the link you provide me, so i suppose M500 can't be performed with this issue.
But the driver was installed cause i used smoothie before. 
It's closed for me,i search solution and i post when i finded in case other people have same issue.

@Bob-the-Kuhn @p3p thank you for your help and patience

Bob-the-Kuhn, 2018-03-15:
Here's what my Win10 device manager says.  Notice that it doesn't mention the Smoothie drive on the com port.

Maybe try the following:
1. uninstall the com port with the smoothie driver on it
2. unplug the RE_ARM
3. reboot
4. plug the RE_ARM back in.
5. see what Windows installs

Or just do a driver update and select the same one as on my PC or on your wife's laptop.  Or just steal her laptop.

[image]
[image]
[image]
[image]

As3uS, 2018-03-15:
@Bob-the-Kuhn  i have the same issue of this thread, but i'am on MKS Sbase ? Using Re-arm driver ?

Bobsy5361, 2018-03-15:
[image]
[New folder.zip](https://github.com/MarlinFirmware/Marlin/files/1816895/New.folder.zip)

Bob-the-Kuhn, 2018-03-15:
Sorry - Marlin uses the same code for all the LPC176x based boards.  Sometimes I forget which motherboard is being used by which person.

Here's more details on the disk drive.

[image]

Here's my files as shown in the snapshot.
[disk.zip](https://github.com/MarlinFirmware/Marlin/files/1816918/disk.zip)

As3uS, 2018-03-15:
Ok my bad ;) so okay for the storage driver, but same issue with 2 computer, one with driver other without.
Think it's not the drivers.

chrisjenda, 2018-03-15:
I will flash the bootloader tonight and try again.  It's very hard for me to work on this issue currently as my main computer's motherboard has decided to take a turn for the worst.

As3uS, 2018-03-18:
Hi there,
@Bob-the-Kuhn @p3p i remove my FullGraphic screen and pluged the usb on my computer and what ? The SD card mounted correctly i can perform a M500 with ponterface. And file eeprom.data write correctly also.

Bob-the-Kuhn, 2018-03-20:
Are you fully functional now?

As3uS, 2018-03-20:
Hi bob, i bought a MKS TFT. 
Now it's fully fonctionnal with a MKS TFT"3.2 Screen but no with the reprap full graphic.
It's ok for me. Thank :)

chrisjenda, 2018-03-20:
Looks like I will be down for another 2 weeks due to the computer problems I am having.  I'm currently using a 10 year old labtop with a 1.3ghz processor while waiting for my Main PC MB to be finish RMA, so bear with me.. lol  I guess you can close this issue and pehaps we could re-open it when I can do more testing, but for now, I am unable to test any suggestions.

closed 2018-04-29 by thinkyhead

Nuttavoot, 2018-12-30:
With the lastest build, I also get "EEPROM datasize error" when issue M500, M501, M504 on RE-ARM board and no file eeprom.data was created on the SD card.