MarlinFirmware/Marlin #10050: [2.0.x] WARNING - Due, native port & Windows - can wipe out bootloader

Bob-the-Kuhn, 2018-03-11:
**The DUE bootloader can get wiped out under the following conditions:**
1. Windows host connected to the native port.
2. Download a new image.

The error message you'll see is the download failed because the com port for the DUE's programming port doesn't exist.

The only way to recover is to re-program the bootloader.

This has apparently been around for a while.  I originally thought it had been introduced by PR #9961 so I reverted it.  I'll revert the revert here shortly.

Bob-the-Kuhn, 2018-03-11:
When the actual download starts the LCD display immediately freezes and the following error messages appear.

**PlatformIO error**
```
Looking for upload port...
Auto-detected: COM4
Forcing reset using 1200bps open/close on port COM4
Uploading .pioenvs\DUE\firmware.bin
No device found on COM4
*** [upload] Error 1
```

**Arduino 1.9 beta build 37 error**
```
Forcing reset using 1200bps open/close on port COM4
C:\Users\bobku\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino/bossac.exe -i -d --port=COM4 -U false -e -w -v -b C:\Users\bobku\AppData\Local\Temp\arduino_build_211226/Marlin_Copy.ino.bin -R 
Send auto-baud
No device found on COM4
Set binary mode
An error occurred while uploading the sketch
```

Bob-the-Kuhn, 2018-03-11:
When the download fails it ends up executing out of the Internal ROM in the 0x0010 0000 region.

After reset in the failing mode the Internal ROM is mapped to the boot area.

After reset in the working mode the FLASH0 is mapped to the boot area.

The FLASH is NOT corrupted.

Bob-the-Kuhn, 2018-03-11:
@ejtagle - ideas?

I don't know if mapping the boot area is a normal part of the download process or not.

Seems like the very beginning of the download sequence actually gets to the SAM3X8E but then the SAM and the ATMEGA16U2 stop talking to each other.

ejtagle, 2018-03-11:
According to section 20 of the datasheet. the SAM-BA bootloader is in ROM, so it is impossible to erase/corrupt it. 
The problem is that the new USB stack is not implementing the Kludge needed to enter the bootloader mode. By kludge, i mean Arduino added an special hook to their (now not used) USB CDC implementation: If you set the Native port to 1200 bauds, and close the port, the SAM3X will reset and enter the bootloader mode.

ejtagle, 2018-03-11:
Without that kludge, it is impossible to enter the programming mode from the native USB port, and thus, it is impossible to reprogram the part. 
I am not using the native port to reprogram the part, so, i never had that issue. I always use the programming port.
But, reimplementing the Arduino Kludge so the native port is usable to program the part is very easy, so i will do that, hoping that it will fix this problem and both ports will be usable for programming purposes.

ejtagle, 2018-03-11:
#10063 should be all what is required to allow programming from the native USB port (tested! - Works for me)

closed 2018-03-15 by Bob-the-Kuhn