MarlinFirmware/Marlin #10053: Heaters turn off at start of print

jamhot1, 2018-03-11:
This a very random problem and seems to pop up without any particular reason. 

I usually issue heating commands to the bed and nozzle so that when I start the print so I can be on top of the 1st layer. as stated, when i get the issue the printer homes and starts a wipe. usually when this happens the printer will switch off the heaters for both bed and nozzle to carry on printing as if all is well. eventually the extruder will stop because of low temperature but it will remain following the toolpath until  it thinks the print is finished. 
[Malin.zip](https://github.com/MarlinFirmware/Marlin/files/1800213/Malin.zip)

thinkyhead, 2018-03-11:
That's pretty strange.

Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

jamhot1, 2018-03-11:
Thank you for the advice. Ok, I tried  the Bugfix and setup accordingly. Still have the same issue.

jamhot1, 2018-03-11:
This is the terminal response when the print starts : 

Send: N5401 G1 X103.025 Y146.038 E4.4659*101
Changing monitoring state from 'Printing' to 'Operational'
Send: G4 P0
Send: ?
Recv:  T:215.10 /0.00 B:56.67 /0.00 @:0 B@:0
Recv: ok N5401 P0 B1
Recv: ok P15 B2
Recv: echo:Unknown command: "?"
Recv: ok P15 B3
Recv:  T:213.33 /0.00 B:56.53 /0.00 @:0 B@:0
Recv:  T:210.83 /0.00 B:56.47 /0.00 @:0 B@:0

Hopefully this may help.

thinkyhead, 2018-03-12:
- Do you see this problem when SD printing?
- Do you see this with all host software?
- Please post the G-code you're printing (ZIP or change extension to .txt).

jamhot1, 2018-03-12:
Right, I think I found the issue and it is related to octoprint rather than Marlin. https://github.com/foosel/OctoPrint/issues/2449
But at least it has given me the chance to try out the newer Marlin. Thank you for your time to reply to this and appreciate all the effort you guys put into this firmware.

closed 2018-03-14 by thinkyhead