MarlinFirmware/Marlin #10059: G02 and G03 strange behaviour

arget888, 2018-03-11:
Hi,

 I bulilded an xy platform with a laser on top for laser engraving. G00 and G01 are working fine and correct but if i enter "G2 I20 J20" it is moving like in the attached picture. I "printed" this with just the one code when the laser was enabled on thermopaper. I am using the latest release 1.1.8 i just added the Full Graphic controller as lcd to my RAMPs 1.4 the rest is the standard configuration of marlin with Arc Support enabled
[image]

thinkyhead, 2018-03-11:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

arget888, 2018-03-11:
Thanks that seems to solve the problem

closed 2018-03-11 by arget888