MarlinFirmware/Marlin #10060: Sixth Extruder

cobramike41, 2018-03-11:
Looking to add a six extruder multi-plexer to my printer and Marlin states it only supports five. Is it possible to add a sixth and if so how?

thinkyhead, 2018-03-11:
Yes. You just have to add the necessary pins, then search all the code for every reference to extruder index 4 and add another instance of the same code, but with E5 in place of E4. It might take a while.

Studiodyne, 2018-03-12:
Make a youtube movie 😄

cobramike41, 2018-03-12:
thinkyhead,

Roger thank you, then I'm assuming that I need to change the checksum to >6 so that i can compile the firmware correct?

Mike

thinkyhead, 2018-03-12:
The test for `EXTRUDERS > 5` would be changed to `EXTRUDERS > 6` as part of your project.

cobramike41, 2018-03-13:
Roger thank you I'll start working on that.

cobramike41, 2018-03-13:
I have a Ramps 1.4 and cnc shield and one thing I noticed was that in the section below of configuration.h, I had to make INACTIVE _EXTRUDER "False" in order to keep the firmware from disabling the 3 steppers on the CNC shield after a few minutes. Basically it would run steppers 3-5 for about 30 secs after I first enabled them then I would have to switch to 1 or 2 (E0, E1) then switch back to 2-4 to get them running again. When I had only had 3 extruders setup with one being on the cnc shield and 2 on the ramps, verses the five I have now, this was not the case.

Have I missed something somewhere or is it because there is only one enable pin for the three steppers on the cnc shield?

Portion of my Config.h:
// @section extruder

#define DISABLE_E false // For all extruders
#define DISABLE_INACTIVE_EXTRUDER false // Keep only the active extruder enabled.

thinkyhead, 2018-03-14:
> is it because there is only one enable pin for the three steppers on the cnc shield?

That would do it. Marlin doesn't make any special consideration for a shared enable pin. It probably should watch out for that.

cobramike41, 2018-03-14:
Thank you again, it's good to know I have a marginal grasp on the firmware. I run octoprint which disable's all the steppers at the end of the print so it shouldn't be an issue. Might be a nice feature though if it could recognize the shared pin and not disable it while any of the attached steppers are printing.

Mike

boelle, 2019-02-20:
@cobramike41 

This Issue Queue is for Marlin bug reports and development-related issues, and we prefer not to handle user-support questions here. (As noted on [this page](https://github.com/MarlinFirmware/Marlin/blob/1.1.x/.github/contributing.md#i-dont-want-to-read-this-whole-thing-i-just-have-a-question).) For best results getting help with configuration and troubleshooting, please use the following resources:

- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- [Tom's 3D Forums](https://discuss.toms3d.org/)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- [Marlin Configuration](https://www.youtube.com/results?search_query=marlin+configuration) on YouTube
- Marlin Discord server. Join link: https://discord.gg/n5NJ59y

After seeking help from the community, if the consensus points to to a bug in Marlin, then you should post a bug report.

cobramike41, 2019-02-20:
Sir,

Understand all, however this really seemed like a bug to me and I did ask on the forums first.

AnHardt, 2019-02-20:
Dear @cobramike41 
All what @boelle wands you to do, is closing this issue if you don't have further questions.
Not touching the issue since about 11 month is a strong indicator for that.

cobramike41, 2019-02-20:
Roger, done.

closed 2019-02-20 by cobramike41