MarlinFirmware/Marlin #10062: Marlin 2.0 bugfix (latest) Re-Arm + TMC2130 problems

gunnarfreyr, 2018-03-11:
So im running Re-Arm + Ramps 1.4sb + TMC2130s from Watterott (all 4 steppers) 

So in SpreadCycle the printer prints ok, but it has some really annoying high pitch sound coming from it and no matter what current i set i still get the sound just a bit diffrent sometimes... From what i understand that has something to do with "chopper frequency" and there should be some way to change that freq. with some TMC_ADV options , but have not figured out how yet so for now Spreadcylcle is useless for me.. 

Stealthchop 
The mode i was hoping to use anyway, so i put it in Stealthchop after the dissapointment with Spreadcycle, but even printing at 35mm/sec with accel. lowered down to 45mm/sec the printer skips layers like a crazy person.. 

Debugging the problem .. 
So i've been trying all kinds of things i've seen over the web, and i've tested the Hybrid mode set to switch over at 40mm/sec (cause the printer always seems to skip layers on travel moves), set all kinds of currents (from 600ma - 1050ma) and i get similar results but after 930 the steppers really quickly warn about overheating so i just stop printing at that point cause usually it's skipped some layers anyways.. 

M122 - The output everybody asks for 
Here is the thing, when i try to compile the firmware with TMC_DEBUG enabled it just will not compile and returns this error below : 

```
Marlin\src\feature\tmc_util.cpp: In instantiation of 'void tmc_status(TMC&, TMC_AxisEnum, TMC_debug_enum, float) [with TMC = TMC2130Stepper]':
Marlin\src\feature\tmc_util.cpp:421:71: required from here
Marlin\src\feature\tmc_util.cpp:388:24: error: second operand to the conditional operator is of type 'void', but the third operand is neither a throw-expression nor of type 'void'
tpwmthrs_val ? SERIAL_ECHO(12650000UL * st.microsteps() / (256 * tpwmthrs_val * spmm)) : SERIAL_CHAR('-');
*** [.pioenvs\LPC1768\src\src\feature\tmc_util.o] Error 1
```

[image]

I have a I3 printer so im not trying to go for some speed records, i would be a happy camper printing at 50-60mm/sec in Stealthchop or Spreadcycle (with out the high pitch noise) and may people seem to be able to do that without any problems.... Any ideas ?

gunnarfreyr, 2018-03-11:
Also wondering .. I noticed in the platformio.ini file 

```
lib_deps =
  https://github.com/MarlinFirmware/U8glib-HAL/archive/dev.zip
  LiquidCrystal_I2C@1.1.2
  TMC2130Stepper
  https://github.com/teemuatlut/TMC2208Stepper/archive/v0.0.3.zip
```
Why is that pointing to TMC2203 but not https://github.com/teemuatlut/TMC2130Stepper ?

teemuatlut, 2018-03-11:
I'll try to get a PR together tomorrow to fix the compile issue. It's only a missing `void` cast for `SERIAL_CHAR`.

Chopper frequency is determined by charge pump voltage, the motor resistance (probably inductance as well), chopper off time and the hysteresis settings. Possibly by the blanking time too.
It'll be very hard to do well suited defaults for the chopper parameters as they depend much on the type of motor that is used and what characteristics it has, as well as the driver voltage.

I'd always recommend adding heatsinks and active cooling for the TMC drivers.

gunnarfreyr, 2018-03-11:
Thank you for the info .. About my problem, i've been doing some experiments and at first i had the heatsinks on the drivers, but no fan cooling them.. And since i was printing from the SD card i didnt have the printer hooked up to my laptop , but when i did that and watched the serial mon. while printing i noticed the steppers reporting that they were close to overheating .. 

And in the TMC2130 datasheet they say : 
"This power dissipation will heat up the PCB cooling
infrastructure also, if operated at an increased duty cycle. This in turn leads to a further increase of
driver temperature. An increase of temperature by about 100°C increases MOSFET resistance by
roughly 50%." 

And from what i understand that means heat = less torque .. So now i added a cooling fan to cool the steppers and that seems to be working .. Cause i've also lowered the current to around 850 for all motors except Z which gets only 750 , and no layershift so far even with the less current , but keeping them cool seems to do wonders for them

annego15, 2018-03-12:
I have the same issue of loosing steps with tmc2130 on the mks sbase. The problem didn't exist in the version 7 days ago. The same gcode didn't loose steps in the previous version, but now it does.

gunnarfreyr, 2018-03-13:
Hey guys, a quick question i've been wondering about .. 

```
  #define R_SENSE           0.11  // R_sense resistor for SilentStepStick2130
  #define HOLD_MULTIPLIER    0.5  // Scales down the holding current from run current
  #define INTERPOLATE       true  // Interpolate X/Y/Z_MICROSTEPS to 256

  #define X_CURRENT          800  // rms current in mA. Multiply by 1.41 for peak current.
  #define X_MICROSTEPS        16  // 0..256

  #define Y_CURRENT          800
  #define Y_MICROSTEPS        16
```

So i've noticed that for example in Tom's video on the 2130s he just uses the "interpolate" for the steps and keeps the microsteps at 16 .. Are there any drawbacks like very low torque if I were to set the microsteps to 256 and Interpolate off ? Or even keep it on for some ultra smooth motion :) ? So insted of some 200 steps pr/mm @ 16 uSteps , use 3200 :) I mean push those mhz in the Re-Arm

boelle, 2019-02-20:
@gunnarfreyr still having problems with latest bugfix 2.0?

gunnarfreyr, 2019-02-20:
> @gunnarfreyr still having problems with latest bugfix 2.0?

hehe no , this is a year old thread :D

boelle, 2019-02-20:
then maybe hit the close botton below?

closed 2019-02-20 by gunnarfreyr