MarlinFirmware/Marlin #10064: After m9939 babystepping stopped working

ruggb, 2018-03-11:
I loaded 12b697c and everything was "perfect" until I started printing and realized babystepping Z is no longer working......
Nothing was changed in the config.
Menu is on LCD; dbl click to BS works; turning the knob increases/decreases the numbers but nothing happens.

thinkyhead, 2018-03-12:
I see no evidence that `BABYSTEPPING` has ever been compatible with CORE kinematics. I'm only surprised that we don't have a sanity-check for it.

thinkyhead, 2018-03-12:
I've now added an implementation for Core-compatible `BABYSTEPPING` in [this branch](https://github.com/thinkyhead/Marlin/tree/bf1_core_babystepping) so that you can enable `BABYSTEP_XY` if you want. Give it a try and see if it does the correct movement in those axes.

ruggb, 2018-03-12:
Does this commit have the Y fix??? I'd rather not move off the BF branch I am on - I confuse easily.........and I don't need BS XY. I never use it. It never worked for Core but I don't have a need for it either. It might be useful on another machine, but it probably worked for those.
Prior to the Y fix babystepping Z worked fine. I didn't change any BS settings. BS Z just stopped working with the Y fix.

xelopics, 2018-03-13:
babysteping works by me not too i have mks sbase on coreXY Custom Build

thinkyhead, 2018-03-14:
> babysteping works by me not too

"Babystepping doesn't work for me either"?

thinkyhead, 2018-03-14:
> Does this commit have the Y fix???

Do you mean "Z fix"? There is no Z fix to the code needed that I can see. Try increasing `MINIMUM_STEPPER_PULSE` as suggested and see if it helps.

thinkyhead, 2018-03-14:
>  I don't need BS XY. I never use it.

No problem, we can find another Core owner to test this for us.

ruggb, 2018-03-14:
When are you going to get BS Z working again????????
When that happens I'll try out XY when you put it into the bugfix branch I am using.
But I still see no need for it on a core machine.

thinkyhead, 2018-03-14:
> When are you going to get BS Z working again????????

That's really obnoxious and childish. If you want help with this you need to chill out.

thinkyhead, 2018-03-14:
Did you try increasing `MINIMUM_STEPPER_PULSE` as I've repeatedly suggested?
You haven't said one way or the other. You simply keep repeating yourself.

I'm starting to think your whole aim is to just troll us.

ruggb, 2018-03-14:
@thinkyhead I went back and reloaded each of the loads. The MINIMUM_STEPPER_PULSE was set to 2 by default but that is specified for 8825 drivers which I have. BS Z started working again - I have no idea why. Nothing changed. But it definitely wasn't functioning previously. I can't imagine a H/W issue as it printed fine. If it quits again, I'll dig into it.
I also loaded ed18791 and the BS X&Y appear to work though they are not available until the print starts. You might want to change that but don't complicate BS Z double click. I can't imagine why one would change X&Y on a Core machine during a print. If used, it is probably only for testing.

Troll indeed! I have nothing better to do.

boelle, 2019-02-20:
@ruggb since it looks like you issue is solved maybe close this one?

ruggb, 2019-02-23:
I don't know if it was solved, but I have totally forgotton about it.

closed 2019-02-23 by ruggb