MarlinFirmware/Marlin #10065: SD card problem

cjake, 2018-03-11:
I can't read SD card.
The log show this message:

16:00:15.882 : start
16:00:15.882 : echo: External Reset
16:00:15.882 : Marlin1.0.2
16:00:15.882 : echo: Last Updated: Dec 17 2016 19:39:29 | Author: (geeetech, I3 config)
16:00:15.882 : Compiled: Dec 17 2016
16:00:15.882 : echo: Free Memory: 3748  PlannerBufferBytes: 1232
16:00:15.882 : echo:Hardcoded Default Settings Loaded
16:00:19.211 : echo:SD init fail
16:00:19.291 : FIRMWARE_NAME:V1.0.03  Sprinter/grbl mashup for gen6 FIRMWARE_URL:https://github.com/Ultimaker/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:I3proB EXTRUDER_COUNT:1 UUID:00000000-0000-0000-0000-000000000000
16:00:19.321 : X:0.00 Y:0.00 Z:0.00 E:0.00 Count X: 0.00 Y:0.00 Z:0.00
16:00:19.321 : echo:Active Extruder: 0
16:00:19.321 : Begin file list
16:00:19.331 : End file list

I changed display circuit with another one but the problem is the same.
Thanks for your help.

thinkyhead, 2018-03-12:
Please test with the latest `bugfix-1.1.x` (and/or `bugfix-2.0.x`) branch to see where it stands. If the problem has been resolved then we can close this issue. If you still see the bad behavior we should investigate further.

cjake, 2018-03-12:
Thanks for reply. I'm a newbie. May you tell me how fix procedure?
Thx a lot.

thinkyhead, 2018-03-12:
Download, configure, and flash [this version](https://github.com/MarlinFirmware/Marlin/archive/bugfix-1.1.x.zip) onto your machine. If you need help with with configuration and troubleshooting, please use the following resources:
- Facebook Group ["Marlin Firmware for 3D Printers"](https://www.facebook.com/groups/3Dtechtalk/)
- RepRap.org [Marlin Forum](http://forums.reprap.org/list.php?415)
- Facebook Group ["Marlin Firmware"](https://www.facebook.com/groups/1049718498464482/)

Phr3d13, 2018-05-03:
@thinkyhead I have 2 machines running the gt2560 board, I could probably maintain most of the Geeetech macnines if you'd like me to fork and make a folder tree of as many of them as I can, I would.

Phr3d13, 2018-05-03:
I meant more like example configs for most/all of the Geeetech machines. I have 4 of them. A Pro W, a Pro C, a Rostock 301, and a Mecreator2.

thinkyhead, 2018-05-04:
@Phr3d13 — We'd be happy to have them in the `example_configurations` folder. The more, the merrier!

closed 2019-07-17 by shitcreek