MarlinFirmware/Marlin #10066: [bugfix-1.1.x] Setting BOARD_RAMBO causes macro expansion warnings

brookstalley, 2018-03-11:
Decided to move to the bugfix branch, but compiling with BOARD_RAMBO is causing macro warnings in the temperature section, which makes me skeptical of using the resulting binary.

Repro:
1) Start with fresh clone of bugfix-1.1.x
2) Edit congfiguration.h to set   #define MOTHERBOARD BOARD_RAMBO
3) Compile

Results:

```
In file included from sketch\fastio.h:61:0,

                 from sketch\MarlinConfig.h:26,

                 from sketch\Marlin.h:35,

                 from sketch\temperature.cpp:27:

sketch\temperature.cpp: In member function 'void Temperature::init()':

sketch\macros.h:106:22: warning: left shift count is negative

 #define _BV(b) (1<<(b))

                      ^

sketch\macros.h:108:24: note: in expansion of macro '_BV'

 #define SBI(n,b) (n |= _BV(b))

                        ^

sketch\temperature.cpp:1103:71: note: in expansion of macro 'SBI'

     #define ANALOG_SELECT(pin) do{ if (pin < 8) SBI(DIDR0, pin); else SBI(DIDR2, pin - 8); }while(0)

                                                                       
sketch\temperature.cpp:1115:5: note: in expansion of macro 'ANALOG_SELECT'

     ANALOG_SELECT(TEMP_0_PIN);

     ^
```

thinkyhead, 2018-03-12:
I'm unable to reproduce your issue. Try closing the Marlin window in Arduino IDE and re-opening the project before re-compiling.

brookstalley, 2018-03-12:
Sigh. I know better. Thanks.

brookstalley, 2018-03-14:
Sadly... it's back, and 99% sure it'll will repro. It was more than just BOARD_RAMBO.

Steps to repro (checked from new git clone and restart of IDE):

1. Start with new git clone of bugfix-1.1
2. In Configuration.h:

- #define MOTHERBOARD BOARD_RAMBO
- #define EXTRUDERS 2
- #define TEMP_SENSOR_0 1
- #define TEMP_SENSOR_1 1
- #define TEMP_SENSOR_2 0
- #define TEMP_SENSOR_3 0
- #define TEMP_SENSOR_4 0
- #define TEMP_SENSOR_BED 1

...will produce the macro expansion warnings listed above.

thinkyhead, 2018-03-14:
Thanks for the update. I'll figure out the issue and patch it soon.

thinkyhead, 2018-03-14:
Ok, so nothing to fix here. Those are just warnings and pertain to code that is stripped from the final binary.

closed 2018-03-14 by thinkyhead