MarlinFirmware/Marlin #10067: M600 - continue print then trigger change

discojon, 2018-03-12:
Is there a way to wait or cache an m600 and continue printing for x mm of filament before triggering the filament change?  My filament sensor is approximately 22" from the nozzle (Taz 5, sensor on filament spinner).  I'd like the sensor to trigger, and the print to continue a defined amount of filament then trigger. 

 If this isn't currently in Marlin I'd like to request it.

Roxy-3D, 2018-03-12:
We could add a #define to delay the trigger of the M600 by the specified amount of filament.

That would be possible.   The problem is finding somebody with time to code it up.

Studiodyne, 2018-03-12:
I love the idea , i have 22" too

Studiodyne, 2018-03-12:
[#9934](https://github.com/MarlinFirmware/Marlin/issues/9934)

tcm0116, 2018-03-13:
I would suggest adding this as a function of the filament monitor and not M600. If you add a delay to M600, then this will cause issues when using M600 to actually change the filament at a certain point in the print, as opposed to changing the filament due to run out.

Studiodyne, 2018-03-13:
@tcm0116 

May be something like this , a real odometer for trigging after a distance
You are good in programmation , it's easy to make , can you make it like this  ?

```
#if defined(M600_TRIG_EXTRUSION_LENGTH)
if (m600_wait_trig_distance == true) {
static float m600_trig_distance = 0.0;
const  float m600_extrusion_count = destination[E_AXIS] - current_position[E_AXIS];
if (m600_extrusion_count) m600_trig_distance += m600_extrusion_count;
if (m600_extrusion_count >= M600_TRIG_EXTRUSION_LENGTH) m600_wait_trig_distance = false;
#endif
```

InsanityAutomation, 2019-04-08:
FYI this was implemented by Marcio and this should no be closed.

guestisp, 2019-04-09:
Dumb question but... usually the filament sensor is very close to the extruder, but above it. If you have the filament sensor at 22'' from the nozzle, but 1'' from the extruder, when the filament sensor says that filament is out, who is pushing the filament to the nozzle? You can still extrude for the distance between the filament sensor and the extruder (in this example, 1''), but in a bowden setup (like you have, if the sensor is 22' above the nozzle) you can't extrude for the whole lenght of the bowden tube....

closed 2019-04-11 by thinkyhead