MarlinFirmware/Marlin #10071: Unable to lit MKS_12864_OLED

soimy, 2018-03-12:
### Description

Unable to lit MKS_12864_OLED with `#define MKS_12864OLED` in Configuration.h.
Branch: latest bugfix-1.1.x
Details: OLED screen will light with bootscreen on re-connected to Serial Port, but will dim immediately after 1 sec. 

### Steps to Reproduce

1. Uncomment `#define MKS_12864OLED`
2. `#define MOTHERBOARD BOARD_MKS_13` (Optional)

**Expected behavior:** OLED screen lit as normal

**Actual behavior:** blank OLED, only light 1s with bootscreen after re-connect via Serial Port.

#### Additional Information

Used to get everything working on version 1.1.0 with patch described at: http://reprap.org/wiki/MKS_12864OLED#Instruction_on_Usage

thinkyhead, 2018-03-12:
It looks like our initialization for this display is not including the 1s delay described in the linked article, even though the pin is labeled…
```cpp
#define LCD_PINS_RS     27 // Pull low for 1s to init
```

Try adding this code to the `lcd_implementation_init` function in `ultralcd_impl_DOGM.h`:
```cpp
#ifdef MKS_12864OLED
  SET_OUTPUT(LCD_PINS_DC);
  OUT_WRITE(LCD_PINS_RS, LOW);
  delay(1000);
  WRITE(LCD_PINS_RS, HIGH);
#endif
```

soimy, 2018-03-12:
This function is implemented in [Marlin_main.cpp](https://github.com/MarlinFirmware/Marlin/blob/9ec8762497547fc9329855ff7f25114622b3fa70/Marlin/Marlin_main.cpp#L13853-L13858) already

soimy, 2018-03-12:
The problem persists, even the bootscreen is gone, only a flash of last info screen before reset.

thinkyhead, 2018-03-12:
> This function is implemented in Marlin_main.cpp already

Ah yes, so it is!

If you go back to 1.1.0 (or any version from 1.1.0 to 1.1.7) the screen works properly?

thinkyhead, 2018-03-12:
Any thoughts on this, @jmdearras?

jmdearras, 2018-03-12:
I will look,  I am running a bugfix 1.1.x from a month ago, and using an MKS-GEN-L currently, and it works.  I'll try upgrading to the latest bugfix and try.

soimy, 2018-03-13:
> If you go back to 1.1.0 (or any version from 1.1.0 to 1.1.8)
does the screen work properly (with the init code and pins added)?
Yes, everything works fine in old 1.1.0-bugfix branch ( Maybe the version half years ago? )
@jmdearras can you give me a copy of your branch? I can test on my machine.

soimy, 2018-03-13:
Turned out the problem is on not proper use of `delay` function makes `LCD_PINS_RS` to be always `LOW`. Just created a PR on that.

boelle, 2019-02-20:
@soimy is your problem solved in latest bugfix 2.0?

soimy, 2019-02-20:
@boelle Yes, works perfectly! Tested on bugfix 2.0 with RAMPS-FD and MKS 12864 OLED

boelle, 2019-02-20:
@soimy then maybe hit the green close botton below?

soimy, 2019-02-20:
Urrrr. Not even noticed it's still open. Sorry.

commit 65ce0303 (2018-03-13): Fix MKS_12864_OLED blank issue

Fixes #10071
closed 2019-02-20 by soimy