MarlinFirmware/Marlin #10072: TMC2208 Stepper driver is disabled using babysteps

Nume1977, 2018-03-12:
When using TMC2208 drivers, without any changes to the chip configuration (dynamic Stealhchop2) and no Uart, if babysteps are used during the print and the value is quickly changed up or down one of the steppers will (randomly) get disabled.
This is a hardware problem, first noticed in klipper, I was now able to reproduce it in Marlin (2.x.x) using the above procedure.

I suspect it is due to the dynamic way the driver changes from stealthChop2 to spreadCycle.
It could either be because the move information hits the threshold of changing the mode but it keeps floating too fast between stealthChop2/spreadCycle or because it hits the sweet spot of the threshold where the driver doesn't know what to do. Either way it will make the driver fail miserably ...

teemuatlut, 2018-03-12:
That should already be handled by the driver. Basically the switching **from** stealthChop **to** spreadCycle is done at a higher speed setting than switching back **from** spreadCycle **to** stealthChop. You can refer to Figure 6.7 in the datasheet to see how it works. In the TMC2208 there are no parameters to change the switching hysteresis value.

thinkyhead, 2018-03-12:
> A possible solution would be…

Indeed, that would not be hard to do if you want to see if it solves the problem.

thinkyhead, 2018-03-12:
Try this in `temperature.cpp`…
```diff
#if ENABLED(BABYSTEPPING)
+ #if ENABLED(HAVE_TMC2208)
+   static bool step_odd;
+   #define DO_BABYSTEP (step_odd = !step_odd)
+ #else
+   #define DO_BABYSTEP true
+ #endif
- LOOP_XYZ(axis) {
+ if (DO_BABYSTEP) LOOP_XYZ(axis) {
    const int curTodo = babystepsTodo[axis]; // get rid of volatile for performance
    if (curTodo) {
      stepper.babystep((AxisEnum)axis, curTodo > 0);
      if (curTodo > 0) babystepsTodo[axis]--;
                  else babystepsTodo[axis]++;
    }
  }
#endif // BABYSTEPPING
```

boelle, 2019-02-20:
@Nume1977 did you try what thinky suggested?

Nume1977, 2019-02-20:
> 
> 
> @Nume1977 did you try what thinky suggested?

Negative, i ended up soldering the TMC2208 bridge and edited the one time programming fuses to enable permanent spreadcycle mode. It is very silent too.

boelle, 2019-02-20:
so this "issue" is not relavant anymore?

boelle, 2019-03-06:
@thinkyhead i think we can close this one

closed 2019-05-16 by thinkyhead