MarlinFirmware/Marlin #10073: UBL on Delta printer

getget1980, 2018-03-12:
### Description
When doing UBL probing on my delta printer, seems like Delta_radius is changed ( ??? ) as the printer does not move anymore into  a X/Y plane but has a convexe movement. (sorry for the English).

Printer usually moves great in X/Y plane (within 0.2mm) when not in UBL mode.

### Steps to Reproduce

M502
M500
G28
G29 P1

**Expected behavior:** [What you expect to happen]
Bed levelling using UBL

**Actual behavior:** [What actually happens]
At the third probing point, probe is too high.

#### Additional Information
Here is configuration .h :
[Configuration.zip](https://github.com/MarlinFirmware/Marlin/files/1801848/Configuration.zip)

A video :
https://www.youtube.com/watch?v=IduaDiqIeM4

Thanks.

getget1980, 2018-03-12:
Yes, 1.1.X (this morning edition :D)

I have updated the config settings.

G33 does not work great (it's worse after calibration ... ) that's why I want to use UBL...
I'm pretty confident on my Delta values (everything was CNC machined) and G33 founds values noticeably far from real ones (Delta_radius 209.78 -> 205.33 ....)

I also noticed than even if my probe is slighly offset from my extruder (X0, Y5.5, Z1.3) and declared as so in configuration.h it seems that it's not taken into account....

Roxy-3D, 2018-03-12:
Wow!   That looks like a nice printer....

The UBL G29 P1 probing is working, right?    If not...   Please post a message and I'll try to help you resolve the issue.   Can you do a UBL G29 P1 and post the mesh map you get?

getget1980, 2018-03-12:
No, G29 P1 does not work (actually it's just what I do on the video).
It probes 1 point near the printer's center, then another one on the left, then lifts up (bowl effect) and probes in the air...

Delta diagonal rod was wrong, I corrected it, G33 now works but G29 no :(

The printer was designed and machined but my students (I'm a teacher :) )

[image]

It has dual printing head with sensor, servo activated :)
[image]

thinkyhead, 2018-03-14:
If you get a chance, also test `AUTO_BED_LEVELING_BILINEAR` just to see if the issue is also there. That will just tell us whether there's a general problem with the way we're dealing with delta kinematics, or whether it's UBL-specific.

getget1980, 2018-03-14:
I will  test this tomorrow. Having problems with G33 too, but i'll start a new issue for G33 ...

G33 makes 2  probings and does not advice values....

18:28:36.716 : echo:Settings Stored (651 bytes; crc 10726)
18:28:36.728 : echo:DEBUG:INFO,ERRORS
18:28:36.728 : echo:Active Extruder: 0
18:28:37.254 : echo:Hardcoded Default Settings Loaded
18:28:39.472 : echo:Settings Stored (651 bytes; crc 10726)
18:28:47.674 : X:0.00 Y:1.00 Z:488.20 E:0.00 Count X:67984 Y:67984 Z:68071
18:28:49.216 : G33 Auto Calibrate
18:28:55.969 : Checking... AC
18:28:55.973 : .Height:488.20  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:209.78
18:28:55.973 : .               Tx:+0.00  Ty:+0.00  Tz:+0.00
18:29:52.540 : Iteration : 01                                std dev:0.618
18:31:14.775 : Calibration OK                                rolling back.
18:31:14.777 : .Height:488.20  Ex:+0.00  Ey:+0.00  Ez:+0.00  Radius:209.78
18:31:14.777 : .               Tx:+0.00  Ty:+0.00  Tz:+0.00
18:31:14.782 : Save with M500 and/or copy to Configuration.h

…

getget1980, 2018-03-19:
Tested Bilinear today.
does not work at all

Video : https://www.youtube.com/watch?v=-PmEegiGzY4

I do homing (ok) then Levelling. Probe deploys ok, and then C Tower (Z axis) moves alone and nothing ?

thinkyhead, 2018-03-21:
That's pretty weird, alright. It's like it's treating the Z movement as if the machine was a Cartesian instead of a Delta, so only the "Z" (C) axis is being moved down instead of doing the kinematic conversion for the move down. The fact that it stops afterward is peculiar. Please do the following so that we can get insight into what's going on there:

- Enable `DEBUG_LEVELING_FEATURE` and re-flash the firmware.
- Connect to your printer from host software such as Cura, Printrun or Repetier Host.
- Issue the command `M111 S247` to enable maximum logging.
- Perform a `G28` to do your standard homing procedure.
- Do `G29` to capture the strange Z movement.
- Copy the log output into a **.TXT file** and attach it to your next reply.

thinkyhead, 2018-03-29:
Can we assume you've solved the issue?

getget1980, 2018-03-29:
Hello, printer is out of order right now (moving to arduino DUE), will post results as soon as it will be operational again.

kubik256, 2018-04-04:
Hi I'm having the same problems on delta with Marlin 2.0.x-bugfix. Auto Bilinear BL and UBL are doing nothing, just strange moves up and down on one tower, until I manualy shut it off.

Printer works fine for smaller objects in the center and has 4 TMC2130 on SPI with StallGuard. Bed needs just a little adjustment for printing larger objects. Inductive probe was manually tested via M119 and works fine. I'm trying to fix this for about a month but still no luck.

My Config files : [Marlin.zip](https://github.com/MarlinFirmware/Marlin/files/1877459/Marlin.zip)

Thanks for any info ;)

thinkyhead, 2018-04-10:
> Inductive probe was manually tested via M119 and works fine.

Do you see the same issue with the current `bugfix-1.1.x`?

kubik256, 2018-04-10:
Not tested :(
I've upgrade printer with probe together with TMC2130. So to test it I have to rebuild printer again and install removed endstops. Or am I wrong and sensorless homing XYZ is now fully working on bugfix-1.1.x?

kubik256, 2018-04-10:
At my delta problem was fixed by using todays bugfix-2.0.x, changing config and one cable :)

Awsome commit here (inverted endstops with TMC2130): https://github.com/MarlinFirmware/Marlin/commit/75037493fa0bb9f62e7d362a1fa783a1e2af7b98

So all endstops was inverted and with combination of loose probe cable, it was quite untraceble problem.

Good luck and many thanks :)

boelle, 2019-02-20:
@getget1980 still having an issue with latest bugfix 2.0?

getget1980, 2019-02-20:
Hello, printer is currently out of service :(

boelle, 2019-03-06:
@getget1980  still out of service?

@thinkyhead i think we can close this one

getget1980, 2019-03-06:
Yes sorry no time to repair it now.

boelle, 2019-03-06:
maybe click close below and open new issue if problem is still there?

getget1980, 2019-03-06:
Ok

closed 2019-03-06 by getget1980