MarlinFirmware/Marlin #10076: [bugfix-1.1.x] X2_CS_PIN?

toenish, 2018-03-12:
I use the latest bug fix  and now Marlin complains about 'X2_CS_PIN':
`'X2_CS_PIN' was not declared in this scope`

I haven't enabled TMC drivers and this pin is nowhere defined. The problem comes in
`#if ENABLED(DUAL_X_CARRIAGE)` in pins.h

thinkyhead, 2018-03-12:
You should always post your Configuration when reporting an issue so we can immediately do a test build. Otherwise it takes us several minutes to follow up, stealing time from other issues.

thinkyhead, 2018-03-12:
I'm unable to reproduce your issue by building the latest bugfix with `DUAL_X_CARRIAGE` and minimal options enabled.

thinkyhead, 2018-03-12:
Issue solved. Will push the update shortly.

closed 2018-03-12 by thinkyhead

thinkyhead, 2018-03-12:
Changes merged. Thanks for the report.