MarlinFirmware/Marlin #10077: Marlin second extruder endtop

xero83, 2018-03-12:
I am mounting a pruse with double extruder, and what I want is to do a homing with 4 points in the bed, two with the extruder on the left, and two with the extruder on the right.
The problem is the configuration or variable that I have to use or modify when telling you that engine 2 is moving at the time of the two points on the right.
Do you know anything about this or where I can find information??

Translated with www.DeepL.com/Translator

thinkyhead, 2018-03-12:
- Homing or leveling?
- Is it a dual extruder with a single carriage?

xero83, 2018-03-12:
It is in homing, I have a double independent extruder and each of the extruders with an endtop.
Sigma type.

thinkyhead, 2018-03-12:
Have you enabled `DUAL_X_CARRIAGE` then? Please ZIP and post your Configuration files.

xero83, 2018-03-12:
Yes, it's activated.
The problem is that I need to use the second extruder when homing at two of the points.

[Uploading test_Marlin-1.0.x.zip…]()

xero83, 2018-03-12:
Yes, it's activated.
The problem is that I need to use the second extruder when homing at two of the points.
[test_Marlin-1.0.x.zip](https://github.com/MarlinFirmware/Marlin/files/1802789/test_Marlin-1.0.x.zip)

thinkyhead, 2018-03-12:
> I need to use the second extruder when homing at two of the points.

So you said. Do you have a second Z probe on the second extruder then?

xero83, 2018-03-12:
Yes; I have a endstop on every extruder.

xero83, 2018-03-12:
How do I make X the second extruder motor?

//EXTRUSOR IZQUIERDO
  #define ABL_PROBE_PT_1_X 50
  #define ABL_PROBE_PT_1_Y 10
  #define ABL_PROBE_PT_2_X 50
  #define ABL_PROBE_PT_2_Y 100
  //ESTRUSOR DERECHO
  #define ABL_PROBE_PT_3_X 205
  #define ABL_PROBE_PT_3_Y 10
  #define ABL_PROBE_PT_4_X 205
  #define ABL_PROBE_PT_4_Y 100

thinkyhead, 2018-03-16:
At this time Marlin doesn't support what you're trying to do, and completely new code will be required. It's a complicated feature because of the many variables involved. So it's unlikely that any of our developers will implement it without reasonable monetary compensation. You may want to find a developer in your local area to work with.

boelle, 2019-02-20:
@xero83 still relavant?

sjasonsmith, 2020-11-10:
It isn't apparent to me what this is actually requesting. It may already be addressed with features in the past 2.5 years. In either case, the conversation has stopped. If work is still needed in this area, it would be good to have an updated request based on Marlin 2.x, which contains more details about the workflow in question.

closed 2020-11-10 by sjasonsmith