MarlinFirmware/Marlin #10078: [2.0.x] Final movement issue

GMagician, 2018-03-12:
I'm facing a weird issue and I think it's related to marlin and not to electronic/mechanical, but I may be wrong.

I'm trying to detect speed/acceleration limits on my printer then I'm using M203 and M201 (at the moment only on X axes) to override firmware limits.
By firmware limits are feed=90mm/s acc=1500mm/s^2
I increased max feed to 110 and using repetier, after home, I've done a G0 X0 and then a G0 X190 using f5400 and everything is ok. Increasing feed to 6600 the whole movement is ok but the stop is abruptly (like acc has been increased but it wasn't).
The weird thing is that if I lower maxacc to 400mm/s^2 motor start ok, move ok but it stop some mm before end (abruptly) and then add a new small movement to reach the end (even with feed 5400).
I have jerk to 15 (or 10 I don't remember now).
This eventing when at home I can make a video to show it and go deep on this with more test.

1) Jerk and acceleration should not be a problem because F5400 moves and stops ok
2) Is not a missing step because problem is when axes is almost at destination, when in full speed it moves ok
3) It shouldn't be a hardware limit because at acceleration/full speed everything is ok (since acc=dec if big gap in decel should also be present during acc)

stepper are controlled by DRV8255/32msteps with Mega2560 and ramps and motor is 400step/round
Anyone has an idea of why I get this ?

thinkyhead, 2018-03-12:
It sounds like Marlin is choking. Try lowering your max feedrates or reducing the number of micro-steps on your main axes.

GMagician, 2018-03-12:
It's not that, 5400 feed moving ok at acc=1500 is not stopping ok with acc=400

GMagician, 2018-03-13:
@thinkyhead I'll close this and do some accurate test.
Just a technical question: how does work the steps generator? I think it's a timer that when elapsed generate a pulse and reinit next elapsed time. Right?
Because if it works this way and issues is because on mega choking, I should see issues at full speed, not when axes is almost stopped, right?

closed 2018-03-13 by GMagician

GMagician, 2018-03-14:
@thinkyhead I'm trying to do some test...is it possible that 'choking' can lead to a board reset after some simple:
G1 X0
G1 X190
repeated many times?

AnHardt, 2018-03-14:
Yes!
If any kind of move-subdivision is involved and the parameters are bad enough.
Please fill a proper issue form, including the configs.

GMagician, 2018-03-14:
@AnHardt I think that 400step/round, 32 microsteps, USB 250KB, REPRAP_DISCOUNT_FULL_GRAPHIC_SMART_CONTROLLER and Mega2560 are enough to choke marlin without bad parameters in cnf...
I think is not a Marlin issue, just too many things to do for a little 2560

AnHardt, 2018-03-14:
Right.
We need the context of **your** problem.