MarlinFirmware/Marlin #10079: Request: on-screen graph of temps for LCD controllers

stibinator, 2018-03-12:
An on-screen temperature graph for LCD graphic screens would be awesome. Basically a menu page with an active graph of any of the heaters, selected by the user.

Thanks

teemuatlut, 2018-03-12:
Wouldn't this after a while just be two horizontal lines?

stibinator, 2018-03-12:
If there was auto-scaling you could dial-in the accuracy, so that as the temperature got to the target the scale of the graph might have the target in the middle with the top and bottom being +/- 1º. Good for testing your PID settings without needing a connected logger.

Not sure that the thermistors have sub-degree accuracy though, so there's that.

thinkyhead, 2018-03-12:
These screens are slooooow. But, it could likely be done with some crafty coding. Temp samples would require minimum 128 bytes of SRAM, pre-allocated. It would be a good entry-level Marlin coding project for somebody who wants to get some embedded C++ practice. Probably won't be implemented before Marlin 2.1.

marcio-ao, 2018-03-12:
`Temp samples would require minimum 128 bytes of SRAM, pre-allocated.`

If you wanted to get fancy, you could pack the vertical component in six bits, which is capable of encoding the Y value from 0 to 63, which is the vertical resolution of the display. This would get you down to 96 bytes. Since the temperature can only change so fast, you could also employ some fancy delta-coding technique to get that storage requirement even lower.

Studiodyne, 2018-03-12:
The first idea , is to make a dynamic database , and , be serious , only 32 bits boards have just a few hours of ressource for this

I suggest it's better to buy , a used LAPTOP and to print by USB , you will have a log , and you can make a graphic with the datas , and some little hours of scripts

But , for all the electronicians here and the 'machine makers ' , I will really enjoy to have this feature , because , to set the perfect ' THERMAL PROTECTION VALUE ',  this feature will make this job
The feature will show a graph , but the stored datas will be used as an ' THERMAL PROTECTION Min/max ' values during a print

Sure , for the moment , and because of the low ressources boards , it's better , to use a laptop and use the log of the host
But it's really a good idea

Studiodyne, 2018-03-12:
@thinkyhead   😄  , don't be so negative with your fans , why 2.1 ??? Are you serious  lolll ? Do we will be alive in ten years to see the 2.1 release  ? loll
And now , i have an idea .... Why do you waste so many time , to clone a second firmware that is the same , and you don't just keep the 2.0 . Now , 2.0 have enough experience to be the only firmware...
So many time wasted , just to rewritte the same code ....
Time is precious ...

Grogyan, 2018-03-12:
My 2cents.
As the displays we normally use are very slow, and if you are wanting graphs as well, why not try to us a cheap mobile phone as a display?

thinkyhead, 2018-03-19:
> So many time wasted

As you are an expert in this, I will take it under advisement.