MarlinFirmware/Marlin #10080: Extruder only turning in one direction

Bricky785, 2018-03-12:
Hello,

I installed Marlin 1.1.8 on a MKS GEN 1.4 with TMC2208 for XYZ and LV8729 for E on my TEVO Tornado. However the extruder is only turning in one direction. I had the jumpers set to 1/16 and 1/32 and that doesn`t change anything. I adjusted the E-Steps and the extruder is extruding the perfect amount of filament. But when it comes to retraction the extruder will turn forward just much faster so that the extruder gear is slipping and grinding into the filament. 

When I try to retract filament with Pronterface the extruder keeps turning in the wrong direction. I plugged the X-Stepper cable into the extruder and the extruder is turning in both directions. 
I can`t find any setting in Marlin that solves this issue. For the XYZ Axis I can invert the endstop but for E there is no endstop. There is only an option to invert the whole axis but that doesnt work. 

Can you tell me whats wrong and which settings do i have to change?

thinkyhead, 2018-03-12:
Are you using `LIN_ADVANCE`?

Bricky785, 2018-03-12:
LIN_ADVANCE is disabled. 

I think I'm already using bugfix-1.1.x because I downloaded the firmware two days ago and the zip is called Marlin 1.1.x.

thinkyhead, 2018-03-12:
"`1.1.x`" isn't "`bugfix-1.1.x`"

https://github.com/MarlinFirmware/Marlin/archive/bugfix-1.1.x.zip

Bricky785, 2018-03-12:
My fault! I will try the bugfix version.

Bricky785, 2018-03-12:
Okay, I took all my settings and put them in the bugfix firmware. Nothing has changed. The extruder is still only turning in one direction.

8Roman8, 2018-03-12:
Have you tried to swap the driver?

8Roman8, 2018-03-12:
not an tmc2208

Bricky785, 2018-03-12:
With a TMC2208 as the driver for the extruder it doesn't move and makes a loud squeaking noise.

8Roman8, 2018-03-12:
Have you checked your cables?

Bricky785, 2018-03-12:
Yes, the cables are fine. Otherwise it also wouldn't turn into the other direction i think.

Bricky785, 2018-03-12:
Thank you very much Bob!

I moved the E-cable to X and it worked fine. Then I moved the LV8729 from E to X, set the jumpers to 1/16 and the stepper only moved in one direction. So it must be a defective driver  I think. I will order a new one and that will hopefully solve this issue.

Thanks to all of you!

closed 2018-03-12 by thinkyhead

reloxx13, 2019-05-02:
@Bricky785  sorry i know this is old, but did a new driver fix it as u suggested?

this actually comes up often in the fb groups, seems like a batch of defective drivers if this is the solution or that those 8729 tend to fail after some time.

Maxracingshox, 2019-06-01:
Two out of three lv8729 I ordered showed the same exact defect as described by the op. I am not even bothering the seller for a refund, but I am pretty pissed tbh. Mostly because I have no spare atm ad this error set me back a lot of time... Well.... the journey is the reward, isn't it? ;)