MarlinFirmware/Marlin #10082: [FR/RFC] Multiple Z axis motors; true bed leveling

marcio-ao, 2018-03-12:
Hello,

We are developing an industrial-class printer where the hot end can move on XY but remains fixed in Z and the bed moves up and down on three independent motors. This allows the bed to be angled in any direction and makes true bed leveling a possibility.

My question is whether Marlin has support for this kind of setup and if not, what it would take to add it. We are nearing the point where we will need to make a choice as to whether Marlin is a viable option for this printer and any feedback would be appreciated.

Thank you,

-- Marcio

thinkyhead, 2018-03-12:
I have implemented this for a machine with four Z motors driving the corners of a large bed. Two boards were involved in the design.

One board acted as the main controller, handling the XYE axes and all the major functions, LCD, etc.

The second board in the project was a RAMPS (in this case) running a very slimmed-down version of Marlin with just the code needed to control the four Z motors, read the 4 opto endstops, and manage procedures like homing and leveling.

The Z-axis controller retained a G-code interpreter for easier control and testing. The Z-axis controller was set up as an I2C slave, while the main board acted as the I2C master. An economical protocol was used to communicate the needed commands to the slave, and to request its status.

I posted the details of the implementation here:
https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879

marcio-ao, 2018-03-12:
@thinkyhead : Two boards -- interesting approach! Our plan was less subtle, we were going to strong-arm Ultimachine into making us a board with a bunch of stepper drivers :)

thinkyhead, 2018-03-12:
That's even better.

marcio-ao, 2018-03-12:
@thinkyhead : If there is an interest in a larger board that would allow this to work, and it wasn't just us, I'm sure it would help make it a reality :)

thinkyhead, 2018-03-12:
There are so many ways to approach machine design nowadays. Whatever works best for your design and keeps complexity down…

Bob-the-Kuhn, 2018-03-12:
The current Marlin has support for two Z motors with the ability to set an offset on one of them when homing.  It moves both Z motors until it hits one of the Z endstops, moves the other until it hits the second Z endstop and then moves the second Z motor by the offset amount.  I'm sure that this can be extended.

Search for DUAL_STEPPER_DRIVERS in Configuration_adv.h for the current options.

There are a few 6 stepper boards already out there.  The 2560 based Azteeg X3 PRO has 8.

A custom shield for the DUE might be of interest.  There's at least 8 unused pins on the RAMPS-FD shield so 8 or 10 drivers is possible.  If you're thinking of using the TMC2130 drivers then the DUE has a internal MUX function that allows an external MUX  to select 1 of 7 or 1 or 15 SPI slaves by using 3 or 4 pins from the processor.

marcio-ao, 2018-03-16:
@Bob-the-Kuhn : Thanks for the info, I'll have to take a look at that feature!

marcio-ao, 2018-08-20:
Looks like this feature may be in the works. https://github.com/MarlinFirmware/Marlin/pull/11062

marcio-ao, 2018-08-20:
Closing this FR since it looks like it may have been implemented

closed 2018-08-20 by marcio-ao

TheLongAndOnly, 2018-08-20:
Hi Marcio,

feel free to contact me, if you have any questions. I'm running this on a DUE+RADDS with DRV8825 Stepper driver.

Holger

marcio-ao, 2018-08-20:
@TheLongAndOnly : The project where we were going to use this was put on hold, but should be restarting sometime in October or so. If I need help then, I'll be sure to reach out to you. Thank you for taking the initiative to get this feature implemented!

jgoo9410, 2019-08-16:
Im currently trying to have marlin communicate the movement data for the Z axis over I2C. I too would like to use the 4 corner method to level the bed using a separate board. Does anyone have the working code?

lanshicai, 2020-03-06:
> I have implemented this for a machine with four Z motors driving the corners of a large bed. Two boards were involved in the design.
> 
> One board acted as the main controller, with a full version of Marlin handling the XYE axes and all the major functions, LCD, serial comms, etc.
> 
> The second board in the project —the Z-axis controller— was a RAMPS (in this case) running a very slimmed-down version of Marlin with just the code needed to control the four Z motors, read the 4 opto endstops, and manage procedures like homing and leveling — or tramming.
> 
> The Z-axis controller retained a G-code interpreter for easier control and testing. The Z-axis controller was set up as an I2C slave, while the main board acted as the I2C master. An economical protocol was used to communicate the needed commands to the slave, and to request its status.
> 
> I posted the details of the implementation here:
> [#4776 (comment)](https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879)

Hello, I want to use 4 Z motors to drive the leveling automatically, but I do n’t have enough knowledge. Can you tell me more about how the two motherboards use I2c to communicate? Thank you.

thinkyhead, 2020-03-06:
> Can you tell me more about how the two motherboards use I2c to communicate?

It's been years since I worked with it, so I have nothing at all to add to https://github.com/MarlinFirmware/Marlin/issues/4776#issuecomment-246262879

Get the code at https://github.com/thinkyhead/Marlin/tree/a3d_main_controller and https://github.com/thinkyhead/Marlin/tree/a3d_zaxis_controller for the actual working implementation.

lanshicai, 2020-03-07:
> Get the code at https://github.com/thinkyhead/Marlin/tree/a3d_main_controller and https://github.com/thinkyhead/Marlin/tree/a3d_zaxis_controller for the actual working implementation.

thank you for your reply.
https://github.com/thinkyhead/Marlin
Does this branch support 4 Z-axis motors without 2 boards?

thinkyhead, 2020-03-07:
> Does this branch support 4 Z-axis motors without 2 boards?

If that's your wish I suggest just waiting for @InsanityAutomation who is now working on extending quad Z to do that trick and maybe a couple of others.

InsanityAutomation, 2020-03-07:
I've got a gtr10 board to test with. Some hardware was delayed shipping from china with the recent issues there. Was hoping to finish it tested before mrrf, but without hardware and time to build it it's looking less likely.

Protean-Man, 2020-06-09:
@InsanityAutomation I am working on something similar but with 3 Z motors. I find the 4th is a little redundant. Any updates on your end with the GTR board?
www.proteusone.net

InsanityAutomation, 2020-06-09:
@Protean-Man 3 motors is done and stable. 4 is mostly done with tweaks and tuning to do, but I havnt had time to assemble hardware around the pandemic. GTR board seems to work well enough though. Mechanical side is whats stopping me atm.

Protean-Man, 2020-06-09:
Thanks for the update. I am still waiting on the gtr board set to test with the 3 motor Z setup. Been a few months now with covid slowdowns. Good to hear the board is working well.