MarlinFirmware/Marlin #10085: LCD timeout turned off after moving axis

marcio-ao, 2018-03-12:
### Steps to Reproduce

It looks like the LCD timeout feature is turned off after a manual axis movement is initiated.

1. Go to the move axis screen
2. Leave the move axis screen
3. The LCD will not timeout to the status screen.

**Expected behavior:** I expect that the LCD will eventually timeout and return to the status screen

**Actual behavior:** It looks like the move axis screen sets "defer_return_to_status" to true, but never clears this value, so the LCD timeout remains off.

thinkyhead, 2018-03-12:
Try this change to `ultralcd.cpp` and see if it solves the issue:
```diff
- void _menu_action_back() { lcd_goto_previous_menu(); }
+ void _menu_action_back() { lcd_goto_previous_menu_no_defer(); }
```

marcio-ao, 2018-03-12:
Yes, this helps. Now if I go two menus back, then the timeout works, so this is an improvement and definitely something that should be fixed.

However, if I just go back once, and leave it on the screen that shows "0.1mm", "1mm", "10mm" then it never times out. The code for the menus is somewhat of a bowl of spaghetti, so I'm not exactly sure where the return to that screen is at.

thinkyhead, 2018-03-12:
I believe that in some earlier version of Marlin `defer_return_to_status` was always set `false` in `lcd_goto_screen` so that every screen must explicitly set it `true` if it wants to hold. I'm not sure when this was altered, but there was some intent there.

thinkyhead, 2018-03-12:
> I'm not exactly sure where the return to that screen is at.

```diff
  void _lcd_move_xyz(const char* name, AxisEnum axis) {
-   if (use_click()) { return lcd_goto_previous_menu(); }
+   if (use_click()) { return lcd_goto_previous_menu_no_defer(); }
```

marcio-ao, 2018-03-12:
That did it, however, the same change needs to be done to:

```
void _lcd_move_e(
    #if E_MANUAL > 1
      int8_t eindex=-1
    #endif
  ) {
if (lcd_clicked) { return lcd_goto_previous_menu_no_defer(); }
```

PS: How did you highlight the diffs like that?

marcio-ao, 2018-03-12:
BTW, when developing the touch UI for Marlin, I came up with what I think is a better way to structure menu code. If you're ever really bored, maybe take a gander at this and see if it is something that could be adapted to make the ultralcd code easier to follow:

https://code.alephobjects.com/diffusion/MARLIN/browse/devel/Marlin/AO_FT810_UI_Screens.h

The key is that I am using "virtual-ish" methods like "onEntry", "onRefresh", "onRefresh" to implement the menus event system. They are not actual virtual functions, because I found out that these are a tremendous memory hog due to the way avr-gcc implements them, but I came up with a clever way to fake my own virtual functions by using a pointer list to static methods in PROGMEM. I think ultralcd tries similar tricks, but the pattern is much more obscure there.

marcio-ao, 2018-03-12:
I confirmed that changing "lcd_goto_previous_menu()" to "lcd_goto_previous_menu_no_defer()" in "_menu_action_back()", "_lcd_move_xyz()" and "_lcd_move_e()" fixes the issue. This ticket can be closed once these changes are made in the code. Thank you!

thinkyhead, 2018-03-14:
> PS: How did you highlight the diffs like that?

When starting a block-quote you can specify the format. For example, ` ```diff `, ` ```cpp `, etc.

thinkyhead, 2018-03-14:
> I came up with what I think is a better way to structure menu code.

A complete redesign of the LCD code is on our TODO list. I'll check out the link and see if it contributes any useful guidance.

closed 2018-03-14 by thinkyhead

marcio-ao, 2018-03-14:
> The current menu system isn't actually that bad, though. It manages to deal with both highlighting and calls to handlers with a single macro per-item, and that's pretty clever in itself.

Well, macros present legibility problems in themselves, when they are nested as deeply as they are in the ultralcd code. That said, I do use macros in my touch UI code myself, so I can't knock them completely :)

thinkyhead, 2018-03-14:
Well, no one has to actually understand the guts of the macros to use them. Behold the simplicity…
```cpp
void _lcd_ubl_output_map() {
  START_MENU();
  MENU_BACK(MSG_UBL_LEVEL_BED);
  MENU_ITEM(gcode, MSG_UBL_OUTPUT_MAP_HOST, PSTR("G29 T0"));
  MENU_ITEM(gcode, MSG_UBL_OUTPUT_MAP_CSV, PSTR("G29 T1"));
  MENU_ITEM(gcode, MSG_UBL_OUTPUT_MAP_BACKUP, PSTR("G29 S-1"));
  MENU_ITEM(function, MSG_UBL_OUTPUT_MAP, _lcd_ubl_output_map_lcd_cmd);
  END_MENU();
}
```

marcio-ao, 2018-03-14:
That's true. But even thought it is obvious in retrospect, it took me quite a while to figure out that the handler for MENU_ITEM(gcode...) was a function named "menu_action_gcode". For a new comer, getting at this by reading the code is challenging.

Another area the macro approach breaks down is that it is difficult to change the behavior of a screen, as such behavior is baked in to the macros. Some of the babystepping screens do not fit the mold, so code like this is exposed:

```
void _lcd_babystep(const AxisEnum axis, const char* msg) {
      if (lcd_clicked) { return lcd_goto_previous_menu_no_defer(); }
      ENCODER_DIRECTION_NORMAL();
      if (encoderPosition) {
        const int16_t babystep_increment = (int32_t)encoderPosition * (BABYSTEP_MULTIPLICATOR);
        encoderPosition = 0;
        lcdDrawUpdate = LCDVIEW_REDRAW_NOW;
        thermalManager.babystep_axis(axis, babystep_increment);
        babysteps_done += babystep_increment;
      }
      if (lcdDrawUpdate)
        lcd_implementation_drawedit(msg, ftostr43sign(planner.steps_to_mm[axis] * babysteps_done));
    }
```

This is called a [leaky abstraction](https://en.wikipedia.org/wiki/Leaky_abstraction). The macros were meant to shield you from all that complexity, but when suddenly they don't work for a particular case, that complexity rises up again and you're left trying to understand the bit that is now hidden behind the abstraction that was meant to protect you.

The situation of making a variant of something that is *almost* like something else crops up quite often in GUI code. Macros don't lend themselves very well to these situations and this is one area in which C++ has better code patterns. I suspect the ultralcd code could use a tad bit less macros and a little more OOP -- the key is striking a good balance between the two.

thinkyhead, 2018-03-15:
Hence the reason I want to redesign the LCD and menu code from the ground up.