MarlinFirmware/Marlin #10087: [FR/Bug] M913 settings is missing in the M503 report

mshicom, 2018-03-12:
### Description
I am using the tmc2130 stepper motor driver and I often rely on M503 to make backups of my machine settings, but I found that the M913 setting for TMC hybrid threshold speed is not shown in the M503 outputs, which I think is kind of a bug.  Hope this can be fixed.

teemuatlut, 2018-03-12:
I'll add "EEPROM support for hybrid threshold" to my growing TODO list.

thinkyhead, 2018-03-14:
@teemuatlut — I'm implementing the EEPROM support. What is the range of valid values for `M913`? I want to make sure the defaults for non-TMC axes are reasonable, or at least obviously impossible. For the moment, I'll just store 0 as the placeholder.

thinkyhead, 2018-03-14:
I noticed in the process of adding `M913` support that the EEPROM is skipping the `M906` output for `IS_TRAMS` in `M503`. That's now fixed as part of the patch.

thinkyhead, 2018-03-14:
Q: Is `SENSORLESS_HOMING` supported also for TMC5130?

teemuatlut, 2018-03-14:
A: Yes.

swilkens, 2020-05-11:
This was provided for in https://github.com/MarlinFirmware/Marlin/pull/10102.

@mshicom or @thinkyhead can close the issue :)

sjasonsmith, 2020-11-10:
I agree with @swilkens that this was included in the referenced PR. That was about 2.5 years ago, so I don't know whether it still works, but if there are issues it should come back as a bug report.

closed 2020-11-10 by sjasonsmith